project-chip/connectedhomeip #18366: [TE9][TC-OO-2.3]Not meeting the expected results

nansenLV, 2022-05-12:
[TC-DM-4.17]step9b step10e step11b step12b step13b step14c step17c step18a step18b step19c step20b step21b step23c step24b doesn't seem to meet the expected results.

!!!EXCEPTED BEHAVIOR:

step9b：
./chip-tool onoff read on-off 1 1

OnOff: TRUE

./chip-tool onoff read on-time 1 1

OnTime: 257

./chip-tool onoff read off-wait-time 1 1

OffWaitTime: 300

step10e：
./chip-tool onoff read on-off 1 1

OnOff: TRUE

./chip-tool onoff read on-time 1 1

 OnTime: 239
 
./chip-tool onoff read off-wait-time 1 1

OffWaitTime: 300

step11b：
./chip-tool onoff read on-off 1 1

OnOff: FALSE

./chip-tool onoff read on-time 1 1

OnTime: 0
 
./chip-tool onoff read off-wait-time 1 1

 OffWaitTime: 236
 
step12b：
./chip-tool onoff read on-off 1 1

OnOff: FALSE

./chip-tool onoff read on-time 1 1

OnTime: 0
 
 ./chip-tool onoff read off-wait-time 1 1

OffWaitTime: 46

step13b：
./chip-tool onoff read on-off 1 1

OnOff:  TRUE

./chip-tool onoff read on-time 1 1

OnTime: 0
 
./chip-tool onoff read off-wait-time 1 1

OffWaitTime: 0

step14c：
[... 304 lines cut ...]
[1652345451.306755][133210:133216] CHIP:EM: Sending Standalone Ack for MessageCounter:9254948 on exchange 28107i

./chip-tool onoff read off-wait-time 6789 1
[1652345466.851846][133221:133227] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_4002 DataVersion: 1330705890
[1652345466.851857][133221:133227] CHIP:TOO:   OffWaitTime: 0
[1652345466.851910][133221:133227] CHIP:EM: Sending Standalone Ack for MessageCounter:10557429 on exchange 9483i

step24b:
./chip-tool onoff read on-off 6789 1
[1652345636.835237][133293:133299] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 1330706197
[1652345636.835263][133293:133299] CHIP:TOO:   OnOff: FALSE
[1652345636.835299][133293:133299] CHIP:EM: Sending Standalone Ack for MessageCounter:6026492 on exchange 46401i

 ./chip-tool onoff read on-time 6789 1
[1652345652.853731][133305:133311] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_4001 DataVersion: 1330706197
[1652345652.853829][133305:133311] CHIP:TOO:   OnTime: 0
[1652345652.853905][133305:133311] CHIP:EM: Sending Standalone Ack for MessageCounter:6002675 on exchange 56372i

./chip-tool onoff read off-wait-time 6789 1
[1652345670.802749][133316:133322] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_4002 DataVersion: 1330706197
[1652345670.803207][133316:133322] CHIP:TOO:   OffWaitTime: 0
[1652345670.803310][133316:133322] CHIP:EM: Sending Standalone Ack for MessageCounter:13436960 on exchange 51563i

chip-tool output:
[tc-oo-2.3.md](https://github.com/project-chip/connectedhomeip/files/8677178/tc-oo-2.3.md)

cjandhyala, 2022-05-24:
@nansenLV , looks like this is your test configuration issue, Can you talk to yinyi.hu@silabs.com and update the ticket

woody-apple, 2022-05-27:
SDK Spec Review: Given this is new a few days old, and tests seem to pass, closing. Please re-open if this issue is still present.

closed 2022-05-27 by woody-apple