project-chip/connectedhomeip #18371: [SQA]-THREAD ConfigStatus Attribute value is remains same (Unchanged).

Survensa, 2022-05-12:
**Problem:**
In TC-WNCV-2.3,As per the test plan if the Write command for mode is passed. The  ConfigStatus value has to changed . B ut it remains the same value . 
If the mode value is 2 then the down-or-close command should fail. But it passes in the thread setup after the successful execution  of write command of mode by value 2.

**Expected behavior:**- Failure (0x01)

**Actual behavior:**-  Success (0x00)

**Steps to reproduce**: `./chip-tool windowcovering down-or-close 1 1`

**Platform :** Thread

**Commit-ID:** https://github.com/project-chip/connectedhomeip/commit/11587f5bf0bb9e621edd4c867c0bc942d804d1ce

**Reference:**  [Test-plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/window_covering.adoc)

**Logs:** 

> [1652356270.423328][31321:31326] CHIP:EM: Removed CHIP MessageCounter:10282731 from RetransTable on exchange 46952i
[1652356270.423399][31321:31326] CHIP:DMG: ICR moving to [ResponseRe]
[1652356270.423499][31321:31326] CHIP:DMG: InvokeResponseMessage =
[1652356270.423557][31321:31326] CHIP:DMG: {
[1652356270.423614][31321:31326] CHIP:DMG: 	suppressResponse = false, 
[1652356270.423725][31321:31326] CHIP:DMG: 	InvokeResponseIBs =
[1652356270.423805][31321:31326] CHIP:DMG: 	[
[1652356270.423864][31321:31326] CHIP:DMG: 		InvokeResponseIB =
[1652356270.423948][31321:31326] CHIP:DMG: 		{
[1652356270.424011][31321:31326] CHIP:DMG: 			CommandStatusIB =
[1652356270.424084][31321:31326] CHIP:DMG: 			{
[1652356270.424157][31321:31326] CHIP:DMG: 				CommandPathIB =
[1652356270.424237][31321:31326] CHIP:DMG: 				{
[1652356270.424317][31321:31326] CHIP:DMG: 					EndpointId = 0x1,
[1652356270.424404][31321:31326] CHIP:DMG: 					ClusterId = 0x102,
[1652356270.424487][31321:31326] CHIP:DMG: 					CommandId = 0x1,
[1652356270.424567][31321:31326] CHIP:DMG: 				},
[1652356270.424656][31321:31326] CHIP:DMG: 				
[1652356270.424729][31321:31326] CHIP:DMG: 				StatusIB =
[1652356270.424810][31321:31326] CHIP:DMG: 				{
[1652356270.424890][31321:31326] CHIP:DMG: 					status = 0x00 (SUCCESS),
[1652356270.424962][31321:31326] CHIP:DMG: 				},
[1652356270.425041][31321:31326] CHIP:DMG: 				
[1652356270.425113][31321:31326] CHIP:DMG: 			},
[1652356270.425195][31321:31326] CHIP:DMG: 			
[1652356270.425260][31321:31326] CHIP:DMG: 		},
[1652356270.425342][31321:31326] CHIP:DMG: 		
[1652356270.425401][31321:31326] CHIP:DMG: 	],
[1652356270.425472][31321:31326] CHIP:DMG: 	
[1652356270.425529][31321:31326] CHIP:DMG: 	InteractionModelRevision = 1
[1652356270.425586][31321:31326] CHIP:DMG: },
[1652356270.425721][31321:31326] CHIP:DMG: Received Command Response Status for Endpoint=1 Cluster=0x0000_0102 Command=0x0000_0001 Status=0x0
[1652356270.425816][31321:31326] CHIP:DMG: ICR moving to [AwaitingDe]
[1652356270.425933][31321:31326] CHIP:EM: Sending Standalone Ack for MessageCounter:1523613 on exchange 46952i

vivien-apple, 2022-05-18:
@jmeg-sfy ?

jmeg-sfy, 2022-05-19:
@Survensa  are you using the latest ToT or are you on TE9 tags ?

Survensa, 2022-05-20:
@jmeg-sfy Faced the same issue 

commit : d346617f388398b40a207c600fb0c0dee3321902 
App : Window app
Connectivity : Thread

woody-apple, 2022-05-23:
Cert Blocker Review: Assigning to myself to follow up to find an owner for window coverings.

kkasperczyk-no, 2022-05-24:
I found the root cause and pushed PR fixing the problem (see above).

Survensa, 2022-05-24:
Thank you so much @kkasperczyk-no will check and update this ticket.