project-chip/connectedhomeip #18374: Messaging protocol active and idle retry intervals are ignored

bluebin14, 2022-05-12:
#### Problem
The active and idle retry intervals defines have no effect other than some mdns text records that also get ignored. E.g. CRI shows as 5000 in mdns but in reality it is about 300. Here's the proof done with chip-tool and a slow responding device.

```
[1652358735.631664][4408:4414] CHIP:IN: Prepared secure message 0xaaaaeb57c588 to 0x0000000000BC614E (2)  of type 0x8 and protocolId (0, 1) on exchange 3334i with MessageCounter:4638975.
[1652358735.631732][4408:4414] CHIP:IN: Sending encrypted msg 0xaaaaeb57c588 with MessageCounter:4638975 to 0x0000000000BC614E (2) at monotonic time: 0000000000DA3B3B msec
...
[1652358735.945655][4408:4414] CHIP:EM: Retransmitting MessageCounter:4638975 on exchange 3334i Send Cnt 0
[1652358735.945820][4408:4414] CHIP:IN: Sending encrypted msg 0xaaaaeb57c588 with MessageCounter:4638975 to 0x0000000000BC614E (2) at monotonic time: 0000000000DA3C75 msec
...
[1652358736.298124][4408:4414] CHIP:EM: Retransmitting MessageCounter:4638975 on exchange 3334i Send Cnt 1
[1652358736.298290][4408:4414] CHIP:IN: Sending encrypted msg 0xaaaaeb57c588 with MessageCounter:4638975 to 0x0000000000BC614E (2) at monotonic time: 0000000000DA3DD6 msec
[1652358736.639063][4408:4414] CHIP:EM: Retransmitting MessageCounter:4638975 on exchange 3334i Send Cnt 2
[1652358736.639235][4408:4414] CHIP:IN: Sending encrypted msg 0xaaaaeb57c588 with MessageCounter:4638975 to 0x0000000000BC614E (2) at monotonic time: 0000000000DA3F2B msec
[1652358737.160186][4408:4414] CHIP:EM: Failed to Send CHIP MessageCounter:4638975 on exchange 3334i sendCount: 3 max retries: 3
```

The 2 relevant defines are:

```
#define CHIP_CONFIG_MRP_DEFAULT_ACTIVE_RETRY_INTERVAL (300_ms32)
#define CHIP_CONFIG_MRP_DEFAULT_IDLE_RETRY_INTERVAL (5000_ms32)
```

In the sample above I set the active interval to 2000, so the entire try and fail cycle should have lasted 5 + 2 + 2 + 2 seconds, instead it failed in less than 2 seconds. Does not seem spec compliant.

 #### Proposed Solution

Do not ignore the active and idle retry intervals.

bluebin14, 2022-05-12:
Clarification: the defines I quoted are the ones in git. For this test I set them differently with active interval 2000. The mdns record was showing the long values.

turon, 2022-05-12:
@bluebin14 thanks for the clarification.  How are you setting up the test such that you expect the first message to be sent with the IDLE interval?

bluebin14, 2022-05-13:
The device was initially using 5000/300. That had the above issue with only 300 being used. Then recompiled  both sides as 5000/2000 -> same thing. Erased KVS -> issue is no longer showing since device is now fast enough.

bluebin14, 2022-05-13:
Apparently the behavior is different on Thread devices, on IP devices I see the exact opposite. At 5000/300 on both sides only 5000 is used:

turon, 2022-07-06:
@bluebin14 was this resolved by #19749?

closed 2022-07-06 by bluebin14