project-chip/connectedhomeip #18379: [TE9][TC-IDM-1.2]Not meeting the expected results

nansenLV, 2022-05-12:
[TC-IDM-1.2]step4 step5 step8 step9  doesn't seem to meet the expected results.

!!!EXCEPTED BEHAVIOR:

step4:

[1650608823029] [89418:1400921] CHIP: [DMG] InvokeResponseMessage =
[1650608823029] [89418:1400921] CHIP: [DMG] {
[1650608823029] [89418:1400921] CHIP: [DMG]         suppressResponse = false, 
[1650608823029] [89418:1400921] CHIP: [DMG]         InvokeResponseIBs =
[1650608823029] [89418:1400921] CHIP: [DMG]         [
[1650608823029] [89418:1400921] CHIP: [DMG]                 InvokeResponseIB =
[1650608823029] [89418:1400921] CHIP: [DMG]                 {
[1650608823029] [89418:1400921] CHIP: [DMG]                         CommandStatusIB =
[1650608823029] [89418:1400921] CHIP: [DMG]                         {
[1650608823029] [89418:1400921] CHIP: [DMG]                                 CommandPathIB =
[1650608823029] [89418:1400921] CHIP: [DMG]                                 {
[1650608823029] [89418:1400921] CHIP: [DMG]                                         EndpointId = 0x1,
[1650608823029] [89418:1400921] CHIP: [DMG]                                         ClusterId = 0x6,
[1650608823029] [89418:1400921] CHIP: [DMG]                                         CommandId = 0x1,
[1650608823029] [89418:1400921] CHIP: [DMG]                                 },
[1650608823029] [89418:1400921] CHIP: [DMG]                                 
[1650608823029] [89418:1400921] CHIP: [DMG]                                 StatusIB =
[1650608823029] [89418:1400921] CHIP: [DMG]                                 {
[1650608823029] [89418:1400921] CHIP: [DMG]                                         status = 0x7e (UNSUPPORTED_ACCESS),
[1650608823029] [89418:1400921] CHIP: [DMG]                                 },
[1650608823029] [89418:1400921] CHIP: [DMG]                                 
[1650608823029] [89418:1400921] CHIP: [DMG]                         },
[1650608823029] [89418:1400921] CHIP: [DMG]                         
[1650608823029] [89418:1400921] CHIP: [DMG]                 },
[1650608823029] [89418:1400921] CHIP: [DMG]                 
[1650608823029] [89418:1400921] CHIP: [DMG]         ],
[1650608823029] [89418:1400921] CHIP: [DMG]         
[1650608823029] [89418:1400921] CHIP: [DMG]         InteractionModelRevision = 1
[1650608823029] [89418:1400921] CHIP: [DMG] },
[1650608823029] [89418:1400921] CHIP: [DMG] Received Command Response Status for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0001 Status=0x7e
[1650608823029] [89418:1400921] CHIP: [TOO] Error: IM Error 0x0000057E: General error: 0x7e (UNSUPPORTED_ACCESS)

step5:

[1651237326.502624][18702:18707] CHIP:DMG: InvokeResponseMessage =
[1651237326.502644][18702:18707] CHIP:DMG: {
[1651237326.502665][18702:18707] CHIP:DMG:         suppressResponse = false, 
[1651237326.502684][18702:18707] CHIP:DMG:         InvokeResponseIBs =
[1651237326.502714][18702:18707] CHIP:DMG:         [
[1651237326.502730][18702:18707] CHIP:DMG:                 InvokeResponseIB =
[1651237326.502761][18702:18707] CHIP:DMG:                 {
[1651237326.502779][18702:18707] CHIP:DMG:                         CommandStatusIB =
[1651237326.502803][18702:18707] CHIP:DMG:                         {
[1651237326.502821][18702:18707] CHIP:DMG:                                 CommandPathIB =
[1651237326.502846][18702:18707] CHIP:DMG:                                 {
[1651237326.502868][18702:18707] CHIP:DMG:                                         EndpointId = 0x0,
[1651237326.502895][18702:18707] CHIP:DMG:                                         ClusterId = 0x30,
[1651237326.502918][18702:18707] CHIP:DMG:                                         CommandId = 0x4,
[1651237326.502940][18702:18707] CHIP:DMG:                                 },
[1651237326.502968][18702:18707] CHIP:DMG:                                 
[1651237326.502984][18702:18707] CHIP:DMG:                                 StatusIB =
[1651237326.503001][18702:18707] CHIP:DMG:                                 {
[1651237326.503019][18702:18707] CHIP:DMG:                                         status = 0x7e (UNSUPPORTED_ACCESS),
[1651237326.503050][18702:18707] CHIP:DMG:                                 },
[1651237326.503079][18702:18707] CHIP:DMG:                                 
[1651237326.503097][18702:18707] CHIP:DMG:                         },
[1651237326.503126][18702:18707] CHIP:DMG:                         
[1651237326.503144][18702:18707] CHIP:DMG:                 },
[1651237326.503172][18702:18707] CHIP:DMG:                 
[1651237326.503188][18702:18707] CHIP:DMG:         ],
[1651237326.503217][18702:18707] CHIP:DMG:         
[1651237326.503234][18702:18707] CHIP:DMG:         InteractionModelRevision = 1
[1651237326.503251][18702:18707] CHIP:DMG: },
[1651237326.503316][18702:18707] CHIP:DMG: Received Command Response Status for Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0004 Status=0x7e
[... 191 lines cut ...]
[1652366249.357405][140652:140657] CHIP:DMG: Received Command Response Status for Endpoint=0 Cluster=0x0000_0003 Command=0x0000_0001 Status=0xc3
[1652366249.357424][140652:140657] CHIP:TOO: Error: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
[1652366249.357433][140652:140657] CHIP:DMG: ICR moving to [AwaitingDe]
[1652366249.357470][140652:140657] CHIP:EM: Sending Standalone Ack for MessageCounter:5287221 on exchange 27825i

step9:
./chip-tool networkcommissioning scan-networks 6789 0
[1652366349.593135][140715:140720] CHIP:DMG:                            StatusIB =
[1652366349.593141][140715:140720] CHIP:DMG:                            {
[1652366349.593157][140715:140720] CHIP:DMG:                                    status = 0x85 (INVALID_COMMAND),
[1652366349.593161][140715:140720] CHIP:DMG:                            },
[1652366349.593204][140715:140720] CHIP:DMG: 
[1652366349.593208][140715:140720] CHIP:DMG:                    },
[1652366349.593225][140715:140720] CHIP:DMG: 
[1652366349.593240][140715:140720] CHIP:DMG:            },
[1652366349.593246][140715:140720] CHIP:DMG: 
[1652366349.593261][140715:140720] CHIP:DMG:    ],
[1652366349.593267][140715:140720] CHIP:DMG: 
[1652366349.593391][140715:140720] CHIP:DMG:    InteractionModelRevision = 1
[1652366349.593410][140715:140720] CHIP:DMG: },
[1652366349.593431][140715:140720] CHIP:DMG: Received Command Response Status for Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0000 Status=0x85
[1652366349.593465][140715:140720] CHIP:TOO: Error: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND)

chip-tool output:
[tc-idm-1.2.md](https://github.com/project-chip/connectedhomeip/files/8680246/tc-idm-1.2.md)

andy31415, 2022-05-20:
@nansenLV - this bug report is somewhat hard to follow because the logs are very long and the verification steps are not easily found (I do not know how to get steps from the code like TC-IDM-1.2). Please help improve the report by adding the following:

- what platform is this tested on? Chip-tool and DUT data
- what are the TC-IDM-1.2 steps ? Link to test plan is ok, a easy to read list of commands would be awesome
- Describe what the expected results are. `Not expected results` does not tell me what is expected nor what it actually gives. I see `INVALID_COMMAND` so assuming that is wrong, but actually have no idea what TC-IDM-1.2 is.

bzbarsky-apple, 2022-05-21:
Test plan is at https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-1-2-invoke-response-action-from-dut-to-th

The failures above in step 4 and step 5 are due to https://github.com/project-chip/connectedhomeip/issues/17221 : we end up doing CASE resumption but the node id has changed, so things break.

For step 8, there  is in fact no identify cluster on endpoint 0; it's on endpoint 1.  So the command being executed is wrong.

For step 9, as of TE9 the spec and impl required a (possibly null) SSID for a WiFi ScanNetworks, but the command is not providing one .  That has since been changed in the spec and impl.

Duplicate of #17221.

closed 2022-05-21 by bzbarsky-apple