project-chip/connectedhomeip #18380: Bugs in resource enforcement for reads/subscribes

bzbarsky-apple, 2022-05-12:
#### Problem
As far as I can tell, we don't properly reserve one read handler _per fabric_.  What we do is:

If a subscribe comes in, and the number of read handlers (so active reads/subscribes) allocated is more than (total available - number of fabrics) we start evicting existing subscriptions.  As a simple example, if MAX_FABRICS were 2 and we supported 3 subscriptions + 1 read per fabric, we would have a cap of 6 subscriptions in `EnsureResourceForSubscription`.  If we have 2 subscriptions on fabric 1, 2 subscriptions on fabric 2, and 2 reads in progress, an attempt to establish a new subscription will evict one of the existing ones.

Worse yet, if we have 3 subscriptions on fabric 1, 2 subscriptions on fabric 2, and 2 reads in progress (total of 7), an attempt to establish a new subscription on fabric 2 will evict _both_ existing subscriptions even though we could have just established our new subscription without evicting anything.

That also means that this block of code in `InteractionModelEngine::OnReadInitialRequest`:
```
    // Reserve the last ReadHandler for ReadInteraction
    if (aInteractionType == ReadHandler::InteractionType::Subscribe && ((handlerPoolCapacity - GetNumActiveReadHandlers()) == 1) &&
        !HasActiveRead())
    {
        ChipLogDetail(InteractionModel, "Reserve the last ReadHandler for IM read Interaction");
        aStatus = Protocols::InteractionModel::Status::ResourceExhausted;
        return CHIP_NO_ERROR;
    }
```
is dead code, because after we finish `EnsureResourceForSubscription` we will always have MAX_FABRICS available read handlers, as far as I can tell.

 #### Proposed Solution
1. Fix logic in `EnsureResourceForSubscription`.  Probably remove the code in `OnReadInitialRequest`, because `EnsureResourceForSubscription` should handle that.  
2. Add tests that would catch this, if we can.  Would involve chunked reads that are still in progress when a subscription request comes in; should be doable with wildcard reads, perhaps.

@erjiaqing @mrjerryjohns

erjiaqing, 2022-05-13:
True

After going through the code again, the issue is 

```
    const auto evictAndUpdateResourceUsage = [&](FabricIndex fabricIndex, bool forceEvict) {
        bool ret           = TrimFabric(fabricIndex, forceEvict);
        usedAttributePaths = mAttributePathPool.Allocated();
        usedEventPaths     = mEventPathPool.Allocated();
        usedReadHandlers   = mReadHandlers.Allocated();
        return ret;
    };
```

Is counting ReadHandlers but it should be usedReadHandlers for Subscriptions since we have `usedReadHandlers < readHandlerCap` below (which should also be `subscriptionCap`)

bzbarsky-apple, 2022-05-13:
Yes, renaming `readHandlerCap` to `subscriptionCap` or better yet `subscriptionHandlerCap` make sense, and replacing all the ` mReadHandlers.Allocated()` with "count the number of live subscription handlers" I think will address the problem...

PR #18533 (2022-05-25): [IM] Fix EnsureResourceForSubscription
#### Problem
`EnsureResourceForSubscription` should only count the resources used by subscriptions, however, it is counting read and subscriptions previously.

Fixes #18380 

#### Change overview

Only count the resources used by subscriptions in EnsureResourceForSubscription.

#### Testing

- Made a "infinity read" transaction when testing the logic of evicting subscriptions.
- The test will fail if the change in src/app/InteractionModelEngine.cpp is reverted.
closed 2022-05-25 by erjiaqing