project-chip/connectedhomeip #18395: TE9 chip tool Darwin testing TC-MF-1.23, 1.24: Getting internal error while pairing when building chip tool Darwin with max fabrics 2

Garik098, 2022-05-12:
As part of the TC-MF-1.23 and TC-MF-1.24, to test the max fabric scenario, the number of fabrics was scaled down to 2 from 16. For chip tool Darwin, used the below command to do that. 

gn gen out/debug --args='chip_mdns="platform" chip_inet_config_enable_ipv4=false target_defines=["CHIP_CONFIG_MAX_FABRICS=2"]'

But when pairing using manual code after the build process, getting chip internal error,

adk@adks-Mac-mini debug % ./chip-tool-darwin pairing manualcode 1 34970112332                                                                           

[1652377522698] [10265:118096] CHIP: [TOO] Running Command
[1652377522698] [10265:118096] CHIP: [DMG] AccessControl: initializing
[1652377522699] [10265:118096] CHIP: [DL] _Init
[1652377522699] [10265:118096] CHIP: [IN] UDP::Init bind&listen port=5541
[1652377522700] [10265:118096] CHIP: [IN] UDP::Init bound to port=5541
[1652377522700] [10265:118096] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1652377522700] [10265:118096] CHIP: [IN] TransportMgr initialized
[1652377522700] [10265:118096] CHIP: [DIS] Init fabric pairing table with server storage
2022-05-12 10:45:22.700 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: g/fidx
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: g/fidx, value FSQABDYBBAEEAgQDGBg=
[1652377522704] [10265:118096] CHIP: [IN] Loading from storage for fabric index 0x1
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/1/r
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/1/r, value FTABAQAkAgE3AyUU0gQkFQEYJgQB++8pJgWBLtErNwYlFNIEJBUBGCQHASQIATAJQQQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFNwo1ASkBGCQCYDAEFKQMV1LA65UJIO4jjtnsQwCMowO8MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtAyHKoUHtwy8UbnFpQ0bTHjnOZEqiP4z9VCLNww1nnprY7bj68YhTmmdXWndsPo4H8a6QOfddgc9t2pLRjg2W2qRg=
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/1/i
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/1/i, value (null)
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/1/n
2022-05-12 10:45:22.704 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/1/n, value FTABAQEkAgE3AyUU0gQkFQEYJgQB++8pJgWBLtErNwYkFQEnERq+gzo14ggwGCQHASQIATAJQQTJiOIGs2+UHflbsVic6fSJ7Hoh35j2RuLrpDNyaEC6HqMnNfzrAR2sWBcVB3JaxZkqB2TEZbEubfpgiUfnp1VXNwo1ASgBGCQCATYDBAIEARgwBBT3BuKzgdS/nQPjxaMxKh2gIiUvWzAFFKQMV1LA65UJIO4jjtnsQwCMowO8GDALQAdDnn/7gT6rEKZ6OZkDVZYc7CRkSHvE8vyPjXpYXxzpVe3U4wni6BTDfLERs8Pnl1wKbczZvgKFmDdvl9IMNNkY
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/1/o
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/1/o, value FSQAATABYQTJiOIGs2+UHflbsVic6fSJ7Hoh35j2RuLrpDNyaEC6HqMnNfzrAR2sWBcVB3JaxZkqB2TEZbEubfpgiUfnp1VXFpT6kO7ueY8sAxG/51bzbO2/zYtd5wdZ60zaKg33uzgY
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/1/m
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/1/m, value FSUA8f8sAQAY
[1652377522705] [10265:118096] CHIP: [IN] Loading from storage for fabric index 0x2
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/2/r
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/2/r, value FTABAQAkAgE3AyUU0gQkFQIYJgRyiwYqJgXyvucrNwYlFNIEJBUCGCQHASQIATAJQQQBuk8Z3QNehN9vCgfrOuxV/fuSx1NyO3AazIhyXGo6tQJ+DMXqQYqMX542rpzHyNLwHolk3hlPB04JkCVXxzgFNwo1ASkBGCQCYDAEFKQMV1LA65UJIO4jjtnsQwCMowO8MAUUpAxXUsDrlQkg7iOO2exDAIyjA7wYMAtANLfzCzMJtC9TKvhnsdiDv79qGs3hqiIyti70LGZPXvKvx1Vyx6jcxt24hSaC8bDdQDy89dBGG1scHP46Nb4V2Rg=
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/2/i
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/2/i, value (null)
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/2/n
2022-05-12 10:45:22.705 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/2/n, value FTABAQEkAgE3AyUU0gQkFQIYJgRyiwYqJgXyvucrNwYkFQInERq+gzo14ggwGCQHASQIATAJQQT03HkcsplHkrkZzrVdHS3JJFM17x4O/tYm+7utj542jASUgFCmZGjyHogXq5/UJzCp98RfEsyzfhCO7avQK/VcNwo1ASgBGCQCATYDBAIEARgwBBRUfcCMi46xgrNUPVuitcCt7QxGozAFFKQMV1LA65UJIO4jjtnsQwCMowO8GDALQBZ/hYq+o6+loc9HmwyRK3Ull+JvNZl1A9l/RyUdZqsUeuuABe5JrpaHniHjZH7l2nCykeDaco1DUqTQuEKJZG8Y
2022-05-12 10:45:22.706 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/2/o
2022-05-12 10:45:22.706 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/2/o, value FSQAATABYQT03HkcsplHkrkZzrVdHS3JJFM17x4O/tYm+7utj542jASUgFCmZGjyHogXq5/UJzCp98RfEsyzfhCO7avQK/VcoYf2Fnri737mmvm7DdByO2KETkIM/PnD0X/Z8qdWJE0Y
2022-05-12 10:45:22.706 chip-tool-darwin[10265:118096] PersistentStorageDelegate Sync Get Value for Key: f/2/m
2022-05-12 10:45:22.706 chip-tool-darwin[10265:118096] CHIPPersistentStorageDelegate Get Value for Key: f/2/m, value FSUA8f8sAQAY
[1652377522706] [10265:118096] CHIP: [TOO] Controller factory startup failed
[1652377522706] [10265:118096] CHIP: [TOO] Run command failure: ../../commands/common/CHIPCommandBridge.mm:49: CHIP Error 0x000000AC: Internal error
[1652377522708] [10265:118096] CHIP: [SPT] VerifyOrDie failure at ../../third_party/connectedhomeip/src/lib/support/Pool.h:317: Allocated() == 0
zsh: abort      ./chip-tool-darwin pairing manualcode 1 34970112332

krypton36, 2022-05-13:
Please try to run `defaults delete com.apple.chiptool `since there appears to be previous fabrics in persistent memory.

bzbarsky-apple, 2022-05-21:
Indeed, if there are >2 fabrics already stored and you rebuild with support for only 2, nothing good can happen....

closed 2022-05-21 by bzbarsky-apple