project-chip/connectedhomeip #18413: [TC-IDM-6.1]DUT fails to responds has UNSUPPORTED_ACCESS, When TH sends invoke message to DUT that it should not have the privilege for the cluster for wildcard event reading.read

Rajashreekalmane, 2022-05-13:
#### Problem
As per Test-Plan, test step 9 says "to read wildcard event Setup the TH, to send invoke request message from TH to DUT  such that it should not have the privilege for the cluster" DUT should responds with  no entry to TH ,but currently DUT responds
 as  
[image]

 * expected behavior:-TH receives Report Data Message with no entry for that event in EventReports list
 * actual behavior:- Operational discovery failed
 * steps to reproduce:- ./chip-tool any  read-event-by-id 0xFFFFFFFF 0xFFFFFFFF 1 0xFFFF --commissioner-nodeid 332211

system configuration
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 11587f5bf0bb9e621edd4c867c0bc942d804d1ce			
Dut:All-cluster-app
 
Test-plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-6-1-events-read-interaction-from-th-to-dut

Spec Reference:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#report-data-action

PFA log below:-
[TC-IDM-13.txt](https://github.com/project-chip/connectedhomeip/files/8684833/TC-IDM-13.txt)

erjiaqing, 2022-05-18:
I found

```
[1652422778.682922][5472:5472] CHIP:DMG: 		EventPath =
[1652422778.682977][5472:5472] CHIP:DMG: 		{
[1652422778.683028][5472:5472] CHIP:DMG: 			Endpoint = 0x0,
[1652422778.683093][5472:5472] CHIP:DMG: 			Cluster = 0x28,
[1652422778.683151][5472:5472] CHIP:DMG: 			Event = 0x0,
[1652422778.683205][5472:5472] CHIP:DMG: 		},
```

Are we really sending wildcard event read requests?

closed 2022-05-18 by vivien-apple

mlepage-google, 2022-05-20:
One more detail on those commands vivien provided: you no longer have to provide the fabricIndex in the ACL entries you write with chip-tool, in fact it is ignored, it will just always use your actual fabricIndex. When you read, you'll see them. So whether you provide them in the write, depends just on whether you find it clearer to have the read and write match exactly, or whether you find it clearer to have the write omit it since it is not used.