project-chip/connectedhomeip #18414: Secure channel [TC-SC-5.2] - Shows error code as unsupported cluster [0xc3]

somu1710, 2022-05-13:
**Problem:**

Secure channel test case [TC-SC-5.2] - While executing the test case it shows error code as unsupported cluster[0xc3] for end point "0" and unsupported access[0x7e] for endpoint "1".

**commands used:**

1. ./chip-tool basic write node-label new 0xffffffffffff0001 0
2. ./chip-tool basic read node-label 1 0
3. ./chip-tool onoff on 0xffffffffffff0001 0
4. ./chip-tool onoff read onoff 1 0
5. ./chip-tool onoff off 0xffffffffffff0001 0

**SS for your reference:**

[image]

[image]

**Logs for your ref:**

[SC-5.2-E0.txt](https://github.com/project-chip/connectedhomeip/files/8685744/SC-5.2-E0.txt)

[SC-5.2 E1.txt](https://github.com/project-chip/connectedhomeip/files/8685746/SC-5.2.E1.txt)

andy31415, 2022-05-13:
@somu1710 what is this test case covering?

There are 2 errors, but 5 commands reported. Which return the errors? Are these the NodeLabel ones?

andy31415, 2022-05-13:
I see the error reported for Cluster 0x06 which is OnOff cluster

sankarselvam, 2022-05-16:
@somu1710 Seems the command & its sequence is wrong, Can you please check with TestPlan  and update the verification steps and then verify this?  If the feature is not developed, this should not be presented in TE9 verification steps sheet.

somu1710, 2022-05-17:
Test plan needs to update on this ticket and it requires feature implementation, we will follow up with the test plan team.