project-chip/connectedhomeip #18417: [TE9][TC-DM-2.1]step1 seems to advertising a short time cause commissioning failed

nansenLV, 2022-05-13:
[TE9][TC-DM-2.1] step1 seems to advertising a short time cause commissioning failed

!!!EXPECTED BEHAVIOR:

sudo ./chip-tool pairing ble-wifi 1 GRLPrivate_EXT matter123 20202021 3840

TH END:
[1641381202.376419][5628:5633] CHIP:CTL: Received success response 0x3df8

sudo ./all-clusters-app --wifi

DUT END:
[1641381202.306840][4431:4431] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)

!!!ACTUAL BEHAVIOR:

./chip-tool pairing ble-wifi 1234 snailRouter 12341234 20202021 3840

THE END:
[1652168928.216654][83370:83374] CHIP:IN: Clearing BLE pending packets.
[1652168928.216705][83370:83374] CHIP:IN: BleConnection Error: ../../examples/chip-tool/third_party/connectedhomeip/src/platform/Linux/BLEManagerImpl.cpp:830: CHIP Error 0x00000032: Timeout
[1652168948.189500][83370:83376] CHIP:SC: PASESession timed out while waiting for a response from the peer. Expected message type was 33
[1652168948.189535][83370:83376] CHIP:TOO: Secure Pairing Failed
[1652168948.189538][83370:83376] CHIP:DIS: Closing all BLE connections
[1652168948.189550][83370:83376] CHIP:TOO: Pairing Failure: ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:275: CHIP Error 0x00000032: Timeout
[1652168948.189832][83370:83370] CHIP:CTL: Shutting down the commissioner
[1652168948.189867][83370:83370] CHIP:CTL: Shutting down the controller
[1652168948.189873][83370:83370] CHIP:CTL: Shutting down the commissioner
[1652168948.189883][83370:83370] CHIP:CTL: Shutting down the controller
[1652168948.189887][83370:83370] CHIP:IN: Expiring all connections for fabric 1!!
[1652168948.189899][83370:83370] CHIP:CTL: Shutting down the commissioner
[1652168948.189908][83370:83370] CHIP:CTL: Shutting down the controller
[1652168948.189910][83370:83370] CHIP:IN: Expiring all connections for fabric 2!!
[1652168948.189915][83370:83370] CHIP:CTL: Shutting down the commissioner
[1652168948.189925][83370:83370] CHIP:CTL: Shutting down the controller
[1652168948.189927][83370:83370] CHIP:IN: Expiring all connections for fabric 3!!
[1652168948.189931][83370:83370] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1652168948.190014][83370:83370] CHIP:DMG: IM WH moving to [Uninitialized]
[1652168948.190018][83370:83370] CHIP:DMG: IM WH moving to [Uninitialized]
[1652168948.190019][83370:83370] CHIP:DMG: IM WH moving to [Uninitialized]
[1652168948.190021][83370:83370] CHIP:DMG: IM WH moving to [Uninitialized]
[1652168948.190024][83370:83370] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1652168948.190035][83370:83370] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1652168948.190141][83370:83370] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-WLWlHh)
[1652168948.190507][83370:83370] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1652168948.190526][83370:83370] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1652168948.190530][83370:83370] CHIP:DL: Inet Layer shutdown
[1652168948.190533][83370:83370] CHIP:DL: BLE shutdown
[1652168948.193740][83370:83370] CHIP:DL: System Layer shutdown
[1652168948.193852][83370:83370] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/PASESession.cpp:275: CHIP Error 0x00000032: Timeout

./chip-lighting-app --ble-device 0

THE END:

[1652168922.690428][24247:24253] CHIP:DL: TRACE: Bluez mainloop starting Thread
[1652168922.692368][24247:24251] CHIP:DL: TRACE: Bus acquired for name MATTER-3840
[1652168922.693016][24247:24251] CHIP:DL: CREATE service object at /chipoble/5eb7/service
[1652168922.695626][24247:24251] CHIP:DL: Create characteristic object at /chipoble/5eb7/service/c1
[1652168922.697732][24247:24251] CHIP:DL: Create characteristic object at /chipoble/5eb7/service/c2
[1652168922.698900][24247:24251] CHIP:DL: CHIP BTP C1 /chipoble/5eb7/service
[1652168922.698978][24247:24251] CHIP:DL: CHIP BTP C2 /chipoble/5eb7/service
[1652168922.698990][24247:24251] CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is FALSE
[1652168922.715860][24247:24247] CHIP:DL: PlatformBlueZInit init success
[1652168922.733047][24247:24247] CHIP:DL: HandlePlatformSpecificBLEEvent 16392
[1652168922.733502][24247:24251] CHIP:DL: BluezPeripheralRegisterAppDone done
[1652168922.733762][24247:24251] CHIP:DL: Create adv object at /chipoble/5eb7/advertising
[1652168922.734692][24247:24251] CHIP:DL: SET service data to {'0xFFF6': <[byte 0x00, 0x00, 0x0f, 0xf1, 0xff, 0x01, 0x80, 0x00]>}
[1652168922.736428][24247:24247] CHIP:DL: HandlePlatformSpecificBLEEvent 16393
[1652168922.736483][24247:24247] CHIP:DL: CHIPoBLE advertising config complete
[1652168922.761589][24247:24251] CHIP:DL: RegisterAdvertisement complete
[1652168922.761894][24247:24247] CHIP:DL: HandlePlatformSpecificBLEEvent 16394
[1652168922.846075][24247:24251] CHIP:DL: Release adv object in BluezAdvertisingRelease

chip-lighting-app output:
[tcdm2.1-ChipLightingAppOutput.md](https://github.com/project-chip/connectedhomeip/files/8685579/tcdm2.1-ChipLightingAppOutput.md)

chip-tool output:
[tcdm2.1-ChipToolOutPut.md](https://github.com/project-chip/connectedhomeip/files/8685585/tcdm2.1-ChipToolOutPut.md)

nansenLV, 2022-05-19:
It seems that the issue is exist in many examples(e.g. chip-tv-app all-cluster-app). I  found the commissionee advertising in a time that so short less than 500ms in the hci log captured by the tool hcidump.

The hci log for device advertising
[image]

cjandhyala, 2022-05-24:
@nansenLV Spec says the adv interval should be b.w. 20 ms and 60 ms.
[image]

nansenLV, 2022-05-25:
@cjandhyala The log captured by hcidump indicate that the time between BLE advertisement begining and ending less than 500 ms, mostly it is just only 1 ms kept for advertising.

andy31415, 2022-05-25:
@nansenLV - could you update the summary a bit:

- provide environment information (what was the DUT env, what chip-tool)
- how often does this reproduce (always or some times only)
- move log dump into an attachment for faster parsing 

See https://github.com/project-chip/connectedhomeip/blob/master/docs/BUG_REPORT.md

woody-apple, 2022-05-25:
Cert Blocker Review: Assigning to @cjandhyala pending new information.

nansenLV, 2022-05-28:
**Chip SDK version**
commit ID : cfc35951be66a664a6efdadea56d1b8ea6e63e96

**DUT(chip-lighting-app) env**
Linux version : 4.4.189 , GCC version : Linaro GCC 6.3-2017.05, Host : aarch64-linux , Cross Compiler : aarch64-linux-gnu-gcc

**chip-tool env**
Ubuntu 20.04 on PC

**How open dose this reproduce**
100%

**How to reproduce**
Step 1: ./chip-lighting-app --ble-device 0
Step 2: ./chip-tool pairing ble-wifi 1111 snailRouter 12345678 20202021 3840
You can do as the steps above, and will see the issue as I said, and the log is same as I have attached.

woody-apple, 2022-06-07:
Cert Blocker Review: @woody-apple to confirm this is a platform specific issue.

Damian-Nordic, 2022-06-09:
@nansenLV I don't have a RPi setup ready at the moment and on my laptop with Ubuntu 22 everything works fine. One thing I noticed is that I don't see the last message from the logs you provided:
```
[1652168922.733762][24247:24251] CHIP:DL: Create adv object at /chipoble/5eb7/advertising
[1652168922.734692][24247:24251] CHIP:DL: SET service data to {'0xFFF6': <[byte 0x00, 0x00, 0x0f, 0xf1, 0xff, 0x01, 0x80, 0x00]>}
[1652168922.736428][24247:24247] CHIP:DL: HandlePlatformSpecificBLEEvent 16393
[1652168922.736483][24247:24247] CHIP:DL: CHIPoBLE advertising config complete
[1652168922.761589][24247:24251] CHIP:DL: RegisterAdvertisement complete
[1652168922.761894][24247:24247] CHIP:DL: HandlePlatformSpecificBLEEvent 16394
[1652168922.846075][24247:24251] CHIP:DL: Release adv object in BluezAdvertisingRelease
```
The message is generate in response to the following signal coming from Bluez:
```
g_signal_connect(adv, "handle-release", G_CALLBACK(BluezAdvertisingRelease), apEndpoint);
```
I also spotted that we set 0 timeout for the advertisement in the code: https://github.com/project-chip/connectedhomeip/blob/master/src/platform/Linux/bluez/Helper.cpp#L183. Perhaps, it's interpreted as "infinity" in newer Bluez versions while your version treats it literally as 0 seconds. Could you try to modify that argument to something bigger like UINT16_MAX and check if that helps?

If not, collecting logs from Bluez may be helpful, too. You can do it by killing Bluez service with `systemctl stop bluetooth` and starting Bluez manually with debug logging enabled: `bluetoothd -d -n`. Then run `chip-lighting-app` again.

nansenLV, 2022-06-17:
@Damian-Nordic   It is really helper for the issue as you said setting UINT16_MAX timeout for advertisement. But there is another issue. It is failed and the message about "no MTU in options in BluezCharacteristicAcquireWrite" feed back.

The log of chip-tv-app is bellow
[no MTU.md](https://github.com/project-chip/connectedhomeip/files/8926697/no.MTU.md)

Damian-Nordic, 2022-06-17:
What operating system is the TV app/RPi run on? It seems it has some pretty old Bluez installed. We can probably investigate further workarounds, but I wonder if it wouldn't be easier to update Bluez on the platform (if we are in contact with the platform-maintainer).

nansenLV, 2022-06-20:
@Damian-Nordic 
**DUT(chip-tv-app) env:**
Linux version : 4.4.189 , GCC version : Linaro GCC 6.3-2017.05, Host : aarch64-linux , Cross Compiler : aarch64-linux-gnu-gcc
**Bluez version:** 
5.50

nansenLV, 2022-06-25:
I have updated the bluez to version 5.5.6 on the DUT platform. It seems that there are no any problems about ble. But it still be commissioned failed. It seems that something wrong about the process of getting network info.

**chip-tool log:**
[chip-tool-TE9.md](https://github.com/project-chip/connectedhomeip/files/8984195/chip-tool-TE9.md)
**chip-tv-app log:**
[chip-tv-app-TE9.md](https://github.com/project-chip/connectedhomeip/files/8984196/chip-tv-app-TE9.md)

Damian-Nordic, 2022-06-27:
Looks like you use a wrong command for commissioning:
```
[1655957039.992946][2523:2530] CHIP:CTL: Parameters supplied: wifi (yes) thread (no)
[1655957039.993753][2523:2530] CHIP:CTL: Device supports: wifi (no) thread(yes)
```

Did you compile the TV app with Thread support? If it's supposed to support WiFi, make sure FeatureMap of NetworkCommissioning cluster is set accordingly.

Damian-Nordic, 2022-06-30:
Depends on what kind of device you want to commission. If you want to commission a Thread device, you use `pairing ble-thread`, if a WiFi device, you use `pairing ble-wifi`. Here the logs suggest that you use `ble-wifi` to commission a Thread device.

cjandhyala, 2022-07-06:
closing per @nansenLV confirmation

closed 2022-07-06 by nansenLV