project-chip/connectedhomeip #18418: [SQA]-THREAD TC-DL-2.14 Unable to execute the credential commands.

Survensa, 2022-05-13:
**Problem:**
In TC-DL-2.14 The Credential command(set,get,clear) are unable to execute in the Chip-tool as we are unable to provide Struct values for Credentials (Credential Type and Credential Index)

**Excepted outcome** :  Success response

**Actual Outcome** : Unable to execute the Command

**Step to reproduce:**  `./chip-tool doorlock set-credential 0 1 2 xxxx 2 0 0 1 1`

**Platform :** RPI - RPI

**Commit** used: 11587f5bf0bb9e621edd4c867c0bc942d804d1ce

**Reference:**  [Test-Plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/door_lock_Cluster.adoc)

**Logs:**

> ./chip-tool doorlock set-credential OperationType Credential CredentialData UserIndex UserStatus UserType node-id/group-id endpoint-id-ignored-for-group-commands [--paa-trust-store-path] [--commissioner-name] [--commissioner-nodeid] [--trace_file] [--trace_log] [--ble-adapter] [--timedInteractionTimeoutMs] [--suppressResponse] [--repeat-count] [--repeat-delay-ms] [--timeout]
[1652348232.057935][30233:30233] CHIP:TOO: Run command failure: ../../commands/common/Commands.cpp:137: Error 0x0000002F

jrhees-cae, 2022-05-20:
The format of the command is incorrect. The Credential parameter needs to be JSON-encoded. For example:
`chip-tool doorlock set-credential 0 '{"credentialType":1, "credentialIndex":1}' '5555' 1 1 0 555 1 --timedInteractionTimeoutMs 1000`

See https://github.com/CHIP-Specifications/chip-test-plans/issues/1298

cjandhyala, 2022-05-24:
@Survensa Can we close this ticket ?

woody-apple, 2022-05-25:
Cert Blocker Review: Assigning to myself and @franck-apple to find an assignee. Tagging @bzbarsky-apple as well

bzbarsky-apple, 2022-05-25:
@Survensa Please attach (not paste) complete logs from the client (including, critically, the exact command being executed) and the server.

My guess is that this is using a credential type not supported by the server app, but the logs would tell for sure.

Survensa, 2022-05-26:
@bzbarsky-apple Here I have attached the log of server and client for this command .
[setcerdential-cntl.txt](https://github.com/project-chip/connectedhomeip/files/8778021/setcerdential-cntl.TXT)
[setcerdential-dut.txt](https://github.com/project-chip/connectedhomeip/files/8778022/setcerdential-dut.txt)

cjandhyala, 2022-05-26:
@jrhees-cae this test is run on nordic nrfdk board

bzbarsky-apple, 2022-05-27:
Yep, that matches my guess from https://github.com/project-chip/connectedhomeip/issues/18418#issuecomment-1137600252

@cjandhyala @Survensa does the feature map for this device claim PIN support?

sethunk, 2022-05-27:
@bzbarsky-apple In lock app, Feature map value: 275 
Binary value : 100010011 ( Which supports PIN,RID,SCH,NOT )

jrhees-cae, 2022-05-27:
FeatureMap value for door-lock example is likely common to all platforms implemented for the example. https://github.com/project-chip/connectedhomeip/blob/master/examples/lock-app/lock-common/lock-app.zap

Therefore, if a certain platform example doesn't support the common features, it would need to override the featuremap value.

franck-apple, 2022-05-27:
Pinging @kkasperczyk-no and @Damian-Nordic

bzbarsky-apple, 2022-05-28:
Yes, sounds like some apps need to set the value at startup, not just use the default from the ZAP file...

kkasperczyk-no, 2022-06-01:
@Survensa @cjandhyala can we close that one? The only issue I see are differences between feature maps vs the real features that are implemented, but it affects probably more examples. If we want to fix this, it could be addressed in a more generic issue that this one.

cjandhyala, 2022-06-01:
per @kkasperczyk-no nordic platform chose not to implement a USR feature, hence this is not a cert blocker

bzbarsky-apple, 2022-06-02:
@cjandhyala Note that the USR feature is required if any of PIN, RID, FPG, or FACE are supported.

kkasperczyk-no, 2022-06-02:
@bzbarsky-apple right, however nRF platform in particular doesn't implement any of those.