project-chip/connectedhomeip #18445: Cannot override DeviceAttestationVerifier in Darwin commissioning flow

tcarmelveilleux, 2022-05-13:
#### Problem
The API surface for Darwin commissioning flows prevents passing a Device Attestation Verifier equivalent that would allow arbitrary policy checks, and allow aggregation/saving of the DAC cert chain and attestation elements/signature, which are needed to do an end-to-end attestation flow if the CA resides off the device (e.g. if running a custom fabric in a multi-fabric situation with app on iOS with Matter framework). Similarly, this prevents having rules that simplify the usage/trust of development device attesattion credentials in a given fabric, where additional rules or validations may be needed.

 #### Proposed Solution

- Allow providing a protocol implementation that is equivalent to `DeviceAttestationVerifier::VerifyAttestationInformation` API surface, if the default policy is insufficient.

bzbarsky-apple, 2022-05-16:
@anush-apple @woody-apple FYI.

terafin, 2022-05-17:
@tcarmelveilleux Can you provide a minimum example API that would fit the needs here?

woody-apple, 2022-05-17:
@tcarmelveilleux Can you provide an example API that would fit the needs here?

woody-apple, 2022-07-19:
@tcarmelveilleux Anything remaining here that's needed? If so, we can move back into 1.0.

woody-apple, 2023-06-23:
@tcarmelveilleux I believe the API needed is now complete, and working, correct? Please re-open if needed.

closed 2023-06-23 by woody-apple