project-chip/connectedhomeip #18455: Thread Tests (nordic board) are failing with Timeout error for chip-tool-darwin

phonnakasturi-apple, 2022-05-14:
Thread Tests (nordic board) are failing with Timeout error for chip-tool-darwin.

When tested against chip-tool they work.

**Steps:**

1. Pair the nordic board with chip tool Darwin
2. Try to read channel attribute.
For instance;
 ./chip-tool-darwin threadnetworkdiagnostics read channel 180 0

Attaching the logs for chip tool darwin:
[chip-tool-darwin_logs.txt](https://github.com/project-chip/connectedhomeip/files/8691890/chip-tool-darwin_logs.txt)

Attaching the logs for chip tool (works):
[chip-tool_logs.txt](https://github.com/project-chip/connectedhomeip/files/8691891/chip-tool_logs.txt)

Damian-Nordic, 2022-05-16:
Looks like Darwin implementation is not able to find an operational service. Do you run both: chip-tool for Linux and macOS against the same Nordic device? If the Linux device is able find the device while Darwin not, it may indicate some issue on the macOS side. Logs from the Nordic board would be useful to rule out SRP issues on the device side.

woody-apple, 2022-06-14:
Assigning to @kean-apple to follow up for logs.

woody-apple, 2022-08-02:
@phonnakasturi-apple can you please re-verify this? If this is still an issue, can you please attach logs from both sides?

closed 2022-08-02 by woody-apple