project-chip/connectedhomeip #18492: [TE9] Get UNSUPPORTED_ATTRIBUTE of read command on DL-3.1

cabin15, 2022-05-17:
#### Problem
Controller: My DUT
Controlee: all-clusters-app of test Harness image on rpi4

When reading attribute of doorlock cluster from all-clusters-app, my DUT got UNSUPPORTED_ATTRIBUTE return.

Belows are the attribute that failed to read.
- number-of-logs-records-supported
- enable-logging
- ledsettings
- local-programming-features
- send-pinover-the-air
- expiring-user-timeout
- alarm-mask
- keypad-operation-event-mask
- remote-operation-event-mask
- manual-operation-event-mask
- rfidoperation-event-mask
- keypad-programming-event-mask
- remote-programming-event-mask
- rfidprogramming-event-mask

Expected result
ReportDataMesage should be displayed like below.
May 11 08:19:15 = chip-service[10172]: ReportDataMessage =
May 11 08:19:15 = chip-service[10172]: {
May 11 08:19:15 = chip-service[10172]:         AttributeReportIBs =
May 11 08:19:15 = chip-service[10172]:         [
May 11 08:19:15 = chip-service[10172]:                 AttributeReportIB =
May 11 08:19:15 = chip-service[10172]:                 {
May 11 08:19:15 = chip-service[10172]:                         AttributeDataIB =
May 11 08:19:15 = chip-service[10172]:                         {
May 11 08:19:15 = chip-service[10172]:                                 DataVersion = 0xc5550562,
May 11 08:19:15 = chip-service[10172]:                                 AttributePathIB =
May 11 08:19:15 = chip-service[10172]:                                 {
May 11 08:19:15 = chip-service[10172]:                                         Endpoint = 0x1,
May 11 08:19:15 = chip-service[10172]:                                         Cluster = 0x101,
May 11 08:19:15 = chip-service[10172]:                                         Attribute = 0x0000_0001,
May 11 08:19:15 = chip-service[10172]:                                 }
May 11 08:19:15 = chip-service[10172]: 
May 11 08:19:15 = chip-service[10172]:                                         Data = 0,
May 11 08:19:15 = chip-service[10172]:                         },
May 11 08:19:15 = chip-service[10172]: 
May 11 08:19:15 = chip-service[10172]:                 },
May 11 08:19:15 = chip-service[10172]: 
May 11 08:19:15 = chip-service[10172]:         ],
May 11 08:19:15 = chip-service[10172]: 
May 11 08:19:15 = chip-service[10172]:         SuppressResponse = true,
May 11 08:19:15 = chip-service[10172]:         InteractionModelRevision = 1
May 11 08:19:15 = chip-service[10172]: }
May 11 08:19:15 = chip-service[10172]: ----------->Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_0001DataVersion: 3310683490
May 11 08:19:15 = chip-service[10172]:   attributeValue: 0

Actual result
May 11 08:28:03 = chip-service[10172]: Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)

Observation:
It is reproduced same using chip-tool on test harness.

bzbarsky-apple, 2022-05-21:
@cabin15 Is this using the door lock app as a server?

bzbarsky-apple, 2022-05-21:
It's probably better to use the example door lock app for testing door lock bits....

cabin15, 2022-05-23:
@bzbarsky-apple 
According to your comment, I did re-test with a few arguments in the below that already failed before, and it still failed with lock-app in the raspberrypi image provided for TE9 with 0xc3 (UNSUPPORTED_CLUSTER).
- number-of-logs-records-supported
- enable-logging
- ledsettings

bzbarsky-apple, 2022-05-24:
@cabin15 Thank you!  Those are definitely wrong: they should either succeed or return UNSUPPORTED_ATTRIBUTE...

cjandhyala, 2022-05-24:
@bzbarsky-apple @andy31415 , the test plan needs to be updated. Not a SDK issue. @cabin15 pls chk with the test plan owner and close the issue.

woody-apple, 2022-05-27:
SDK Spec Review: Assigning to @bzbarsky-apple to update lock items.

closed 2022-06-02 by andy31415