project-chip/connectedhomeip #18494: [TE9] Get UNSUPPORTED_COMMAND of invoke command on DL-3.2

cabin15, 2022-05-17:
#### Problem
Controller: My DUT
Controlee: all-clusters-app of test Harness image on rpi4

1. When sending invoke command of doorlock cluster to all-clusters-app, my DUT got UNSUPPORTED_COMMAND return.
Belows are the step that failed due to UNSUPPORTED_COMMAND.
- Step 4, 5, 6, 7, 8, 9, 10, 13, 16, 17, 18, 19, 20, 21, 22, 23, 24, 26

`May 11 17:27:09 = chip-service[10987]: Error: IM Error 0x00000581: General error: 0x81 (UNSUPPORTED_COMMAND)`

2. When sending invoke command of doorlock cluster to all-clusters-app, my DUT got NOT_FOUND return.
Belows are the step that failed due to UNSUPPORTED_COMMAND.
- Step 11, 14

`May 11 17:33:48 = chip-service[10987]: Error: IM Error 0x0000058B: General error: 0x8b (NOT_FOUND)`

3. When reading attribute of doorlock cluster from all-clusters-app, my DUT got INVALID_COMMAND return.
Belows are the step that failed due to UNSUPPORTED_COMMAND.
- Step 32, 33

`May 11 19:33:18 = chip-service[11270]: Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)`

bzbarsky-apple, 2022-05-21:
It's probably better to use the example door lock app for testing door lock bits....

cabin15, 2022-05-23:
@bzbarsky-apple 
According to your command, I re-tested this using lock-app. 
It still shows same UNSUPPORTED_COMMAND error

`ubuntu@ubuntu:~/apps$ sudo ./chip-tool doorlock get-log-record 1 1 1
[1653349066.886976][127206:127206] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1653349066.887510][127206:127206] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1653349066.887671][127206:127206] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1653349066.887781][127206:127206] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1653349066.888117][127206:127206] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-UwqoHs)
[1653349066.888722][127206:127206] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1653349066.888776][127206:127206] CHIP:DL: NVS set: chip-counters/reboot-count = 3 (0x3)
[1653349066.889601][127206:127206] CHIP:DL: Got Ethernet interface: eth0
[1653349066.890309][127206:127206] CHIP:DL: Found the primary Ethernet interface:eth0
[1653349066.891083][127206:127206] CHIP:DL: Got WiFi interface: wlan0
[1653349066.891171][127206:127206] CHIP:DL: Failed to reset WiFi statistic counts
[1653349066.891260][127206:127206] CHIP:IN: UDP::Init bind&listen port=0
[1653349066.891415][127206:127206] CHIP:IN: UDP::Init bound to port=36523
[1653349066.891445][127206:127206] CHIP:IN: BLEBase::Init - setting/overriding transport
[1653349066.891472][127206:127206] CHIP:IN: TransportMgr initialized
[1653349066.891535][127206:127206] CHIP:DIS: Init fabric pairing table with server storage
[1653349066.891692][127206:127206] CHIP:IN: Loading from storage for fabric index 0x1
[1653349066.894680][127206:127206] CHIP:IN: Loading from storage for fabric index 0x2
[1653349066.895418][127206:127206] CHIP:IN: Loading from storage for fabric index 0x3
[1653349066.896098][127206:127206] CHIP:DIS: Add fabric pairing table delegate
[1653349066.899698][127206:127206] CHIP:ZCL: Using ZAP configuration...
[1653349066.903869][127206:127206] CHIP:DL: Avahi client registered
[1653349066.905118][127206:127206] CHIP:CTL: System State Initialized...
[1653349066.905212][127206:127206] CHIP:CTL: Setting attestation nonce to random value
[1653349066.905262][127206:127206] CHIP:CTL: Setting CSR nonce to random value
[1653349066.905368][127206:127206] CHIP:CTL: Setting attestation nonce to random value
[1653349066.905418][127206:127206] CHIP:CTL: Setting CSR nonce to random value
[1653349066.906323][127206:127206] CHIP:CTL: Generating NOC
[1653349066.907240][127206:127206] CHIP:DIS: Verifying the received credentials
[1653349066.908966][127206:127206] CHIP:DIS: Added new fabric at index: 0x1, Initialized: 1
[1653349066.909081][127206:127206] CHIP:DIS: Assigned compressed fabric ID: 0x1CFA0B1314F90229, node ID: 0x000000000001B669
[1653349066.909107][127206:127206] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x1CFA0B1314F90229
[1653349066.909201][127206:127206] CHIP:CTL: Setting attestation nonce to random value
[1653349066.909257][127206:127206] CHIP:CTL: Setting CSR nonce to random value
[1653349066.910045][127206:127206] CHIP:CTL: Generating NOC
[1653349066.911179][127206:127206] CHIP:DIS: Verifying the received credentials
[1653349066.913296][127206:127206] CHIP:DIS: Added new fabric at index: 0x2, Initialized: 1
[1653349066.913340][127206:127206] CHIP:DIS: Assigned compressed fabric ID: 0xD29F9E1A03ABC930, node ID: 0x000000000001B669
[1653349066.913370][127206:127206] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0xD29F9E1A03ABC930
[1653349066.913483][127206:127206] CHIP:CTL: Setting attestation nonce to random value
[1653349066.913551][127206:127206] CHIP:CTL: Setting CSR nonce to random value
[1653349066.914469][127206:127206] CHIP:CTL: Generating NOC
[1653349066.915569][127206:127206] CHIP:DIS: Verifying the received credentials
[1653349066.917705][127206:127206] CHIP:DIS: Added new fabric at index: 0x3, Initialized: 1
[1653349066.917748][127206:127206] CHIP:DIS: Assigned compressed fabric ID: 0xD49E990DB6CCC7FE, node ID: 0x000000000001B669
[1653349066.917777][127206:127206] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0xD49E990DB6CCC7FE
[1653349066.950047][127206:127211] CHIP:DL: CHIP task running
[1653349066.950316][127206:127211] CHIP:DL: HandlePlatformSpecificBLEEvent 32787
[1653349066.950350][127206:127211] CHIP:TOO: Sending command to node 0x1
[1653349066.950382][127206:127211] CHIP:CSM: FindOrEstablishSession: PeerId = 1CFA0B1314F90229:0000000000000001
[1653349066.950408][127206:127211] CHIP:CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
[1653349066.950451][127206:127211] CHIP:DIS: Resolving 1CFA0B1314F90229:0000000000000001 ...
[1653349066.952230][127206:127211] CHIP:DL: Avahi resolve found
[1653349066.952306][127206:127211] CHIP:DIS: Node ID resolved for 1CFA0B1314F90229:0000000000000001
[1653349066.952338][127206:127211] CHIP:DIS:     Addr 0: [fe80::870:daaf:910d:24ba]:5540
[1653349066.952586][127206:127211] CHIP:DIS: UDP:[fe80::870:daaf:910d:24ba%eth0]:5540: new best score: 3
[1653349066.952613][127206:127211] CHIP:DIS: Checking node lookup status after 2 ms
[1653349066.952634][127206:127211] CHIP:DIS: Keeping DNSSD lookup active
[1653349067.149928][127206:127211] CHIP:DIS: Checking node lookup status after 200 ms
[... cut ...]
[1653349067.197205][127206:127206] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1653349067.197291][127206:127206] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1653349067.197327][127206:127206] CHIP:DL: Inet Layer shutdown
[1653349067.197356][127206:127206] CHIP:DL: BLE shutdown
[1653349067.197385][127206:127206] CHIP:DL: System Layer shutdown
[1653349067.197542][127206:127206] CHIP:TOO: Run command failure: IM Error 0x00000581: General error: 0x81 (UNSUPPORTED_COMMAND)
ubuntu@ubuntu:~/apps$`

bzbarsky-apple, 2022-05-24:
`get-log-record` is a command that is only supported if the `LOG` feature is supported.  Is that feature claimed to be supported in this case?

cabin15, 2022-05-26:
@bzbarsky-apple 
As no guide is described on test-plan for enabling `LOG` feature on server, I couldn't get to it.
BTW, as there has been a lot of changes on doorlock test-plan, let me re-test DL-3.2 according to latest test-plan.
Let me close this.
Thank you.

closed 2022-05-26 by cabin15