project-chip/connectedhomeip #18496: [TE9] Test plan does not describe the verification steps how to simulate the event on DL-3.3

cabin15, 2022-05-17:
#### Problem
On DL-3.3, test steps asks me to do "Simulate a xxxxx Event by setting the AlarmCode to 0", such as

"Simulate a DoorLockAlarm Event by setting the AlarmCode to 0"
"Simulate a DoorStateChange event by setting the DoorState to 2"
"Simulate a LockOperation event by setting OperationSource to 2"
"Simulate a LockUserChange event by changing the Credential"

As no verification step is described how to achieve this, I'm not fully able to perform the test for DL-3.3.

bzbarsky-apple, 2022-05-21:
This looks like an issue for the test plan repo at https://github.com/CHIP-Specifications/chip-test-plans/

@cjandhyala

closed 2022-05-21 by bzbarsky-apple

bzbarsky-apple, 2022-05-21:
Filed as https://github.com/CHIP-Specifications/chip-test-plans/issues/1311