project-chip/connectedhomeip #18508: [SQA] TC-GR-2.2 Chiptool allows Group ID ' 0 ' in Group Cluster

Survensa, 2022-05-17:
#### Problem
During the execution of AddGroup Command with the GroupID as 0 we should get response as INVALID_COMMAND as the Group ID should be in the inclusive range of 1 to 65535

 * expected behavior : Response should be INVALID_COMMAND
 * actual behavior : Getting Response as status: 135 (Status is enum8 type)
 * Command used : `./chip-tool any command-by-id 0x0004 0x00 '{ "0": 0, "1": "my-group-name"}' 1 0`
 * system configuration : Platform - RPI, Network - Ble-wifi
 * Log : [Groups_Issue_TH.txt](https://github.com/project-chip/connectedhomeip/files/8709558/Groups_Issue_TH.txt)

cjandhyala, 2022-05-17:
@Survensa Pls provide spec reference as well here.

bzbarsky-apple, 2022-05-20:
The spec is not self-consistent, right?  On the one hand, it lists a constraint, which should result in INVALID_COMMAND per IM definitions of this stuff.  On the other hand, the "Effect on Receipt" prose says:
> The server verifies that the GroupID field contains a valid group identifier in the range 0x0001 - 0xffff. If the GroupID field contains a group identifier outside this range, the status SHALL be INVALID_VALUE and the server continues from step 6.

The implementation seems to be returning INVALID_VALUE, correct?

@Rob-Houtepen-Signify @CamWms could you please take a look at the spec end here?

bzbarsky-apple, 2022-05-20:
And for `AddGroupIfIdentifying` it's even worse: per IM that should error out even if not identifying, but per prose an invalid GroupID is ignored there if not identifying...

woody-apple, 2022-05-23:
Cert Blocker Review: Assigning to @bzbarsky-apple to update when the related spec issue is addressed

bzbarsky-apple, 2022-05-25:
Spec PR merged.  SDK now matches spec.

bzbarsky-apple, 2022-05-25:
@cjandhyala @Survensa The test plan needs to be updated to the spec change.

bzbarsky-apple, 2022-05-25:
Actually, reopening, since the spec got aligned on INVALID_COMMAND but the SDK does INVALID_VALUE.

bzbarsky-apple, 2022-05-26:
Spec issue for this: https://github.com/CHIP-Specifications/connectedhomeip-spec/issues/5242

Spec PR for that issue: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5253

bzbarsky-apple, 2022-06-02:
Actually, reading the spec more carefully there is still a mismatch between IM and groups cluster there: the groups cluster treats the contraint violation as a _success_ for purposes of the IM.

bzbarsky-apple, 2022-06-02:
OK.  That spec change merged.  I created https://github.com/project-chip/connectedhomeip/pull/19124 to address an issue in the groups cluster that I found while looking into this, but otherwise this is done.

The test plan needs to be updated to expect the right behavior: a _successful_ command response, that has an error in the "Status" field.

closed 2022-06-02 by bzbarsky-apple