project-chip/connectedhomeip #18520: EFR32MG24 lighting-app create Hardfault

cjszz2001, 2022-05-17:
#### Problem
<EFR32MG24 lighting-app create Hardfault at startup.
Build the lighting-app on the EFR32_BOARD=BRD4186A
on the branch of interop_testing_2022_01_18.

Here is the catpure of the J-link RTT log:
00> <efr32 > ==================================================
00> <efr32 > chip-efr32-lighting-example starting
00> <efr32 > ==================================================
00> <efr32 > Init CHIP Stack
00> <detail> [-] Initializing key value store
00> <detail> [-] KeyValueStore init complete: active keys 0, deleted keys 0, sectors 4, logical sector size 8192 bytes
00> <info  > [DL] Setting device name to : "EFR32_LIGHT"
00> <efr32 > Initializing OpenThread stack
00> <info  > [DL] OpenThread started: OK
00> <info  > [DL] Setting OpenThread device type to ROUTER
00> <efr32 > Starting Platform Manager Event Loop
00> <efr32 > Starting OpenThread task
00> <efr32 > Starting App Task
00> <efr32 > Starting FreeRTOS scheduler
00> <info  > [DL] Bluetooth stack booted: v3.2.1-b216
00> <info  > [DL] RAIL version:, v2.11.1-b0
00> <detail> [DL] Thread task running
00> <detail> [DL] CHIP task running
00> <detail> [DL] Start BLE advertissement
00> <info  > [ZCL] Using ZAP configuration...
00> <info  > [DL] _OnPlatformEvent default:  event->Type = 32780
00> <detail> [DL] OpenThread Sta<efr32 > Turning light ON
00> <efr32 > Light ON
00> <info  > [ZCL] On/Off set value: 1 1
00> <info  > [ZCL] Toggle on/off from 0 to 1
00> <info  > [ZCL] On Command - OffWaitTime :  0
00> <info  > [ZCL] Cluster callback: 0x0000_0006
00> <info  > [ZCL] On/Toggle Command - Stop Timer
00> <info  > [ZCL] Cluster callback: 0x0000_0006
00> <info  > [SVR] Saved into server storage: a/1/6/0
00> <info  > [ZCL] Cluster callback: 0x0000_0006
00> <info  > [SVR] Saved into server storage: a/1/8/0
00> <info  > [ZCL] Cluster callback: 0x0000_0008
00> <info  > [ZCL] Level Control attribute ID: 0x0000_0000 Type: hu Value: 32, length 1
00> <error > HardFault:
00> 
00> SCB->CFSR   0x00008200
00> SCB->HFSR   0x40000000
00> SCB->MMFAR  0x00000000
00> SCB->BFAR   0x00000000
00> SP          0x20006c68
00> R0          0x00000000
00> 
00> R1          0x20006c40
00> 
00> R2          0x00000000
00> 
00> R3          0x00000000
00> 
00> R12         0x00000003
00> 
00> LR          0x08019c3b
00> 
00> PC          0x08023bac
00> 
00> PSR         0x81100000>
 #### Proposed Solution
<the light example should run properly>

bzbarsky-apple, 2022-05-25:
@cjszz2001 Is this the same issue as #18522?

cjszz2001, 2022-05-25:
Flash the bootloader and it boots up properly.
you can find the sample bootloader file for BRD4186 in matter repo: "connectedhomeip\third_party\efr32_sdk\repo\platform\bootloader\sample-apps\bootloader-storage-spiflash-single\efr32mg24a010f1536gm48-brd4186a"

andy31415, 2022-05-30:
@cjszz2001 could you test this on ToT master?

interop_testing_2022_01_18 is a very old branch by chip development speed standards.

andy31415, 2022-05-30:
I built Tot and flashed:

```
./out/efr32-brd4161a-light/chip-efr32-lighting-example.flash.py --erase --verbose --reset
```

and it booted fine without a hardfault.

closed 2022-05-30 by andy31415