project-chip/connectedhomeip #18532: [TE9][TC-SC-5.1] Lock-app is failing on step 2 (./chip-tool groupkeymanagement key-set-write)

epeng100, 2022-05-18:
In [TC-SC-5.1] Step 2,
./chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42,
"groupKeySecurityPolicy": 0, "epochKey0":
"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1":
"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2":
"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 1 0

The result shows UNSUPPORTED_CLUSTER for lock-app.

Checking examples/lock-app/lock-common/lock-app.zap, there is no "Group Key Management" cluster as other applications.
If the test is mandatory for all type devices, lock-app should support the cluster.

cjandhyala, 2022-05-24:
@epeng100 is there a reason why you need to run this TC on lock app?

epeng100, 2022-05-25:
No, we tested it because the case is in TEDS list for lock app.

bzbarsky-apple, 2022-06-10:
This cluster is mandatory for all devices.

That said, devices are allowed to support only one keyset per fabric (for the IPK) if they don't implement the Groups cluster.  And locks are absolutely allowed (and in fact strongly encouraged not to) implement the Groups cluster.  So doing that keyset write is absolutely allowed to fail for a lock, just not with UNSUPPORTED_CLUSTER.

woody-apple, 2022-06-15:
Spec Issue Review: Sankar to confirm that this test case is indeed running and passing.

sankarselvam, 2022-06-16:
@woody-apple  This test is not passing in QA. This test has been linked to the testplan issue  [1453](https://github.com/CHIP-Specifications/chip-test-plans/issues/1453)  
Test plan requires an update as well.

Morozov-5F, 2022-06-16:
> > does that mean we should enable the groups-cluster in the lock-app and configure it properly?
> 
> 
> 
> Group key management, not groups.

Yep, sorry, I meant exactly that, thanks

PR #19684 (2022-06-23): Update lock-app ZAP file to contain Group Key Management cluster and expose mandatory attribute in the Identify cluster
#### Problem
* Fixes #18532 by adding Group Key Management cluster
* Fixes #19181 by enabling all the mandatory features in the lock-app ZAP file

#### Change overview
* ZAP changes for the lock-app, update auto-generated files

#### Testing
* Run commands from #18532 
* Run linter from #19181
closed 2022-06-23 by woody-apple