project-chip/connectedhomeip #18545: Thermostat cluster "SetpointRaiseLower" command shwoing status = 0xc3 (UNSUPPORTED_CLUSTER)

manjunath-grl, 2022-05-18:
Summary:
[TSTAT-2.2] Thermostat cluster  "SetpointRaiseLower" command shwoing status = 0xc3 (UNSUPPORTED_CLUSTER) when executed 
Spec screenshot:
[image]
Testplan screenshot:
[image]
Execution logs:
[setRaisePoint.txt](https://github.com/project-chip/connectedhomeip/files/8714260/setRaisePoint.txt)
[image]

bzbarsky-apple, 2022-05-25:
This command is not enabled in all-clusters-app, looks like.

woody-apple, 2022-05-27:
SDK Spec Review: Tagging this as a potential cert blocker, tagging @cjandhyala

cjandhyala, 2022-05-30:
closing per @manjunath-grl request

cjandhyala, 2022-06-01:
@manjunath-grl per QA team you need to run this test using the thermostat app.

manjunath-grl, 2022-06-15:
@cjandhyala  tried in TOT its passing
    SHA used: https://github.com/project-chip/connectedhomeip/commit/c1e1d018820c0fdd51f124f607b9207f31fae2ff
    Client app: Chip-tool
    Server app: thermostat app
Execution logs:
[TC_TSTAT_2_2.zip](https://github.com/project-chip/connectedhomeip/files/8906488/TC_TSTAT_2_2.zip)

closed 2022-06-15 by manjunath-grl