project-chip/connectedhomeip #18580: Shutdown and reclaim server-side IM resources on underlying session going away

mrjerryjohns, 2022-05-18:
#### Problem

IM resources on the server today continue to exist and linger around even if their bound session has gone away.

#### Proposal

Register for the `OnSessionReleased` delegate in the IM engine perhaps, and iterate over all Read/Command/Write handlers and evict those that are bound to the matching session.

kghost, 2022-05-27:
Read/Command/Write commands are bound to `ExchangeContext`, there are no resource bound to `OnSessionReleased`, such that nothing should be handled inside `OnSessionReleased`.

The subscription has something related to session, where should be removed when `OnSessionReleased` is triggered. Synced with @yunhanw-google that even if the resource is not cleared in `OnSessionReleased`, the resources will be released later. On the client side, the subscription will be released when there is no report for a while; On the server side, an error raised by `ReadHandler::SendReportData` will remove the subscription.

Handles `OnSessionReleased` in `ReadHandler` is better-to-have, to free resources as soon as possible.

yunhanw-google, 2022-05-27:
per chat with @kghost , with his below change, when report messaged fails to send with crmp retires, session would not be released and would be kept alive. And Session would be released only when there is no enough resource in session pool. It seems this feature would not be that useful since Session would be kept alive for long time.

https://github.com/project-chip/connectedhomeip/pull/18107
https://github.com/project-chip/connectedhomeip/pull/18878
@bzbarsky-apple @mrjerryjohns

bzbarsky-apple, 2022-06-13:
@mrjerryjohns When a session is actually released, we are going to close the relevant exchanges.  So we don't need to listen for OnSessionReleased; we just need to listen for OnExchangeClosing, right?

mrjerryjohns, 2022-06-23:
> So we don't need to listen for OnSessionReleased; we just need to listen for OnExchangeClosing, right?

For interactions that always have an open exchange, yes, this is true. For subscriptions specifically, this won't suffice.

mrjerryjohns, 2022-06-23:
Nonetheless, I think the utility of this is diminished since:

1. Sessions won't really go away unless evicted to make way for a new one.
2. If new sessions get created from the same peer, the subscription is likely to shift to the new one anyways.

Will see if this crops up as an issue during inter-op.

yunhanw-google, 2022-08-12:
@mrjerryjohns  is this related to this ticket https://github.com/project-chip/connectedhomeip/issues/21084?

mrjerryjohns, 2022-08-15:
No, that one is about deleting IM objects on fabric deletion. This is about deleting objects on session deletion..

bzbarsky-apple, 2023-02-13:
I think we need a clear idea of what's left to do here, which cases would be helped but this, etc, broken out by interaction type.

And perhaps separate issues by interaction type, since I expect the constraints and tradeoffs to be different...