project-chip/connectedhomeip #18583: Shift multi-callback management from `OperationalDeviceProxy` to `CASESessionManager`

mrjerryjohns, 2022-05-18:
#### Problem

Comments like [this](https://github.com/project-chip/connectedhomeip/pull/18569#discussion_r876335578) one reveal that clean-up of `OperationalDeviceProxy` instances on failure is just fraught with peril. Currently, callers of `FindOrEstablishSession` are responsible for releasing `OperationalDeviceProxy` instances on error. This is really error-prone. The ability for `OperationalDeviceProxy` to support multiple registered callbacks on the same instance complicates this story.

#### Proposal

Extending the idea in that comment, up-level the multi-callback management scheme in `OperationalDeviceProxy` to `CASESessionManager`. This simplifies `OperationalDeviceProxy` down to just managing a single callback, while permitting `CASESessionManager` to now take-over the responsibility of free'ing up the proxy instance on error, while still retaining the multi-callback capability of the high-level `FindOrEstablishSession` API.

mrjerryjohns, 2022-05-18:
FYI @msandstedt @bzbarsky-apple

bzbarsky-apple, 2022-05-19:
So the CASESessionManager would need to keep track of which callbacks go with which proxy, right?  But yes, we could do that...

msandstedt, 2022-05-19:
We use the SecureSessionTable and Sessions directly, so aren't exposed to any of this.  I defer to others on how `OperationalDeviceProxy` and `CASESessionManager` should work together.

mrjerryjohns, 2022-06-23:
Dupe of #19259