project-chip/connectedhomeip #18586: [TE9] Test plan does not describe the verification steps how to simulate the failure on TC-DIAG-TH_NW-3.5

cabin15, 2022-05-19:
#### Problem
On TC-DIAG-TH_NW-3.5, test steps (Step 1~8) asks me to do "Simulate a xxxxx failture", such as

"Simulate an Unspecified failure on TH"
"Simulate a LinkDown failure on TH"
"Simulate a HardwareFailure on TH"
"Simulate a NetworkJammed Failure on TH"
"Simulate an Unspecified failure on TH"
"Simulate a LinkDown failure on TH"
"Simulate a HardwareFailure on TH"
"Simulate a NetworkJammed Failure on TH"

As no verification step is described how to achieve this, I'm not fully able to perform the test for TC-DIAG-TH_NW-3.5

bzbarsky-apple, 2022-05-21:
This looks like an issue for the test plan repo at https://github.com/CHIP-Specifications/chip-test-plans/

@cjandhyala

bzbarsky-apple, 2022-05-21:
Filed as https://github.com/CHIP-Specifications/chip-test-plans/issues/1310

closed 2022-05-21 by bzbarsky-apple