project-chip/connectedhomeip #18590: [TC-DL-2.11] ClearYearDay Schedule Command is giving UNSUPPORTED_COMMAND Error

kowsisoundhar12, 2022-05-19:
ClearYearDay Schedule Command is giving( UNSUPPORTED_COMMAND 0x81 ) Error
[image]
[image]
[Test_TC_DL_2_11.zip](https://github.com/project-chip/connectedhomeip/files/8723257/Test_TC_DL_2_11.zip)

andy31415, 2022-05-20:
ClearYearDaySchedule has conformance `SCH`. Could you validate if the door lock feature map claims this feature is supported?

kowsisoundhar12, 2022-05-31:
@andy31415  SCH bit is not set in feature map

DanM-AssaAbloy, 2022-08-04:
What door lock example app was this performed on?
Does this still apply to the latest version of the sample app?

kowsisoundhar12, 2022-11-28:
this issue has been fixed ,hence closing this issue

closed 2022-11-28 by kowsisoundhar12