project-chip/connectedhomeip #18592: need to pairing again by chiptool after ubuntu restared

Leo2442926161, 2022-05-19:
#### Problem
<what's wrong or missing, please include any applicable:
 * expected behavior
 commissioner and commissionee are able to interaction with each other everytime since paired.

 * actual behavior
 it lost connection when I restarted ubuntu, I have using "sudo ./chip-tool pairing ble-thread" to pair it again for establish a connection.

 * steps to reproduce
 just reboot ubuntu in VMware.

 * system configuration
 ubuntu 20.04

bzbarsky-apple, 2022-05-21:
@Leo2442926161 Did you pass `--KVS` arguments to the relevant applications so that their persistent storage files would go somewhere that's not `/tmp`?  Because if not, then an OS restart would clear `/tmp` and hence delete the storage, so the "I am commissioned" information would be lost.

Leo2442926161, 2022-05-21:
Hi bzbarsky, thanks for your help.

how to add --KVS arguments to relevant applications?

bzbarsky-apple, 2022-05-21:
You just do it:
```
chip-all-clusters-app  --KVS /home/whatever/something
```

Leo2442926161, 2022-05-21:
Hi bzbarsky, thanks again,

I am using chiptool, there no --KVS parameter

bzbarsky-apple, 2022-05-23:
@Leo2442926161 When you say "I restarted ubuntu", is the thing running on ubuntu chip-tool or the server side?

Leo2442926161, 2022-05-24:
@bzbarsky-apple the chip-tool run on ubuntu

bzbarsky-apple, 2022-11-30:
chip-tool supports setting `TMPDIR` now to the value of your choice.

closed 2022-11-30 by bzbarsky-apple