project-chip/connectedhomeip #18595: [TC-DL-2.12&TC-DL-2.16] Following commands are missing in Door lock cluster

kowsisoundhar12, 2022-05-19:
Following commands are missing in door lock cluster
GetUserStatus
SetRFIDCode
GetRFIDCode

[image]
[image]
[execution-logs.zip](https://github.com/project-chip/connectedhomeip/files/8725323/execution-logs.zip)
[exe-logs.zip](https://github.com/project-chip/connectedhomeip/files/8725324/exe-logs.zip)

bzbarsky-apple, 2022-05-25:
In the device library the USR feature is _required_ for Matter, and therefore these commands are _required_ to not be supported in a Matter device.

bzbarsky-apple, 2022-05-25:
As in, those commands are not part of Matter at all.  They're in the cluster definition, but Zigbee-only.

closed 2022-05-25 by bzbarsky-apple