project-chip/connectedhomeip #18601: [TC-IDM-5.2]-Chip-tool needs to provide a way to set delayed write request to verify the step 3

Rajashreekalmane, 2022-05-19:
**#### Problem**
As per Test-Plan, [TC-IDM-5.2] test step 3 expecting that we should set "Wait for 5 seconds(Timer has expired) and then send the Write Request Message to the DU", But Chip-tool needs to provide a way to set wait time for write request messages.

**PFA screenshot below:-**
[image]

**Test-Plan screenshot:-**
[image]

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#timed-write-transaction

bzbarsky-apple, 2022-05-21:
There is YAML syntax for this, but not command-line syntax....

woody-apple, 2022-05-23:
Cert Blocker Review: Assigning to @cjandhyala to follow up, looks like the test should be runnable in YAML right now.

Rajashreekalmane, 2022-05-25:
I tried in chip-tool interactive mode.

steps to reproduce:-
sudo ./chip-tool interactive start
then at the shell prompt  sent the timed request 
./chip-tool administratorcommissioning open-basic-commissioning-window 500 1 0 --timedInteractionTimeoutMs 1000
waited for 5 mins
then sent a write command ./chip-tool levelcontrol write on-level 1 1 1
getting response as success instead of unsupported access.

Shall I know, is there any way to verify this test step in chip-tool?

bzbarsky-apple, 2022-05-25:
Just to repeat:

> There is YAML syntax for this, but not command-line syntax....

woody-apple, 2022-05-27:
@cjandhyala Can you please close this, and file a ticket to update the test plan?

sankarselvam, 2022-06-08:
@vivien-apple We still see this issue and not able to send a delayed request. Hence we reopened the ticket for the further follow-up.

Rajashreekalmane, 2022-06-09:
When it was tried in another setup, it worked fine.

PR #18999 (2022-06-02): [chip-tool] Add --repeat-count and --repeat-delay-ms to write commands
#### Problem

Add `--repeat-count` and `--repeat-delay-ms` to write commands for `chip-tool`

Fix #18601

@bzbarsky-apple Similar to the YAML way. This is not exactly doing what the test step says in the sense that we don't really know if the "status response message" has been received but in practice it does the job...

#### Change overview
 * Add `--repeat-count` and `--repeat-delay-ms` to write commands for `chip-tool`

#### Testing

I tried to do:
`./out/debug/standalone/chip-tool testcluster write timed-write-boolean f [... cut]
closed 2022-06-09 by Rajashreekalmane