project-chip/connectedhomeip #18608: [TC-IDM-2.2]-Chip-tool needs to support the "SuppressResponse field" value to set for Read Request Message to the DUT to read an attribute

Rajashreekalmane, 2022-05-19:
#### Problem
Chip-tool needs to provide API to set the SuppressResponse field value.
As per the Test-Plan, the test step 32 in [TC-IDM-2.2] is expecting to set SuppressResponse field set to False.

**PFA screenshot below:-**
[image]

**system configuration**
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 25e241e
Dut: All-Clusters-app

**Test-Plan Screenshot:-**
[image]

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#outgoing-read-request-action

bzbarsky-apple, 2022-05-22:
This test plan is very confused; there is no "SuppressResponse" field in a Read Request.

Filed https://github.com/CHIP-Specifications/chip-test-plans/issues/1313 to get the test plan fixed to make sense.

bzbarsky-apple, 2022-05-22:
@cjandhyala

closed 2022-05-22 by bzbarsky-apple