project-chip/connectedhomeip #18616: [SU-2.2] While updateNotAvaliable (delay 60), OTA-R does not send query image after 2 minutes

doublemis1, 2022-05-19:
#### Problem
After waiting for 2 minutes DUT does not retry, it only retries with -q busy, not with -q updateNotAvailable

[provider_su22_3.log](https://github.com/project-chip/connectedhomeip/files/8730182/provider_su22_3.log)

 #### Proposed Solution
<suggested fix, suggested enhancement>

andy31415, 2022-05-20:
@doublemis1 - please provide spec information about why expected spec behaviour is not met.

@cjandhyala is this a cert blocker? I seem to remember some slack discussions saying that on "update not available" we currently wait one day (i.e. saying not available now, try later).

Generally trying to update every 2 minutes even when the other side says "not available" sounds reasonably wrong to me. Would I want all my IOT devices (I should have 10s easily, maybe 100) to try and check for updates every 2 minutes? that sounds painful. Or maybe I misunderstand the test case.

bluebin14, 2022-05-20:
@andy31415 ignore "2 minutes", the title should be "When updateNotAvailable, OTA-R does not send query image after DelayedActionTime". Also "retry after x minutes" does not mean "retry every x minutes forever". The suggested number of retries is at most 2. See also https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5248

woody-apple, 2022-05-23:
Cert Blocker Review: Assigning to @carol-apple, given this is OTA. Can you please update next steps here? Thanks!

woody-apple, 2022-05-26:
Cert Blocker Review: @cjandhyala Can you please follow up on the test plan here?

carol-apple, 2022-05-27:
The current behavior is spec compliant. Please note that there is a pending spec update which should clarify the language on this: https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5248

To summarize, when a `NotAvailable` status response is received, it means there is no update available. Regardless of whether there is a `DelayActionTime`, the spec does not state that a retry must occur.

ukatira, 2022-05-27:
Issue tracking test plan update based on pending spec update - https://github.com/CHIP-Specifications/chip-test-plans/issues/1380

franck-apple, 2022-05-27:
No action needed on the SDK side. Closing.

bluebin14, 2022-05-28:
Closing as per https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5248

closed 2022-05-28 by bluebin14