project-chip/connectedhomeip #18633: Update AddNOC to follow final error handling spec text

tcarmelveilleux, 2022-05-19:
#### Problem
In https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5231, error handling got normalized to follow reality. The updates need to be taken in once landed in spec. This was a Must-Fix 1.0 spec issue

 #### Proposed Solution
- Rename `MissingAdminNode` enum value to `InvalidAdminSubject`
- Rename `CaseAdminNode` command field to `CaseAdminSubject`
- Remove `MissingIpk` enum value

andy31415, 2022-05-26:
@cjandhyala @tcarmelveilleux @bzbarsky-apple  - is this actually a cert-blocker?

The way I read it, these are enum value renames ... do these enum values show up in cert tests? they sound internal to me, however I am not familiar with them.

woody-apple, 2022-05-26:
Cert Blocker Review: @cjandhyala is updating the test plan to fully cover the test, which will make this a cert blocker.

bzbarsky-apple, 2022-05-27:
@andy31415 These are spec-defined enums; they are not internal, but sent on the wire.  And it's not just renames, it's also which is returned when.

PR #19819 (2022-06-24): Implement shadow fail-safe data in FabricTable
#### Problem
- FabricTable management during commissioning did not properly
  handle the fact that committing needs to only be done on
  CommissioningComplete, which prevented the AddTrustedRootCertificate,
  UpdateNOC and AddNOC command semantics to be implemented properly
  and prevented proper state observation of operational credential
  clusters server

Fixes #7695
Issue #8905
Fixes #18633
Issue #17208
Fixes #15585

#### Change overview
- Removes direct access to FabricInfo from everywhere, which caused
  possibly stale FabricInfo references  [... cut]
commit e5e09f5c (2022-06-24): Implement shadow fail-safe data in FabricTable (#19819)

* Implement shadow fail-safe data in FabricTable

- FabricTable management during commissioning did not properly
  handle the fact that committing needs to only be done on
  CommissioningComplete, which prevented the AddTrustedRootCertificate,
  UpdateNOC and AddNOC command semantics to be implemented properly
  and prevented proper state observation of operational credential
  clusters server

Fixes #7695
Issue #8905
Fixes #18633
Issue #17208
Fixes #15585

This PR:

- Removes direct access to FabricInfo from everywhere, which caused
  p [... cut]
closed 2022-06-24 by tcarmelveilleux