project-chip/connectedhomeip #18634: Chip tool Darwin testing: As part of Multi Fabrics tests, reading fabrics output not matching chip tool output

Garik098, 2022-05-19:
As part of Multi Fabric tests, there are few steps to read fabrics after remove commissioner steps too see if the respective fabric was removed. In chip tool Darwin, read fabrics command is not giving the list of fabrics at the end to see if the respective fabrics has been removed while chip tool gives that list. 

Need this list to conform if the fabric removed was as per the expectation.  

**Chip tool Darwin reading fabrics command:** 
./chip-tool-darwin operationalcredentials read fabrics 1 0 --fabric-filtered 0
**Output**:
[1652983963053] [77495:4815111] CHIP: [EM] Received message of type 0x5 with protocolId (0, 1) and MessageCounter:1630642 on exchange 61488i
[1652983963053] [77495:4815111] CHIP: [EM] Found matching exchange: 61488i, Delegate: 0x7fe86b61ebb0
[1652983963053] [77495:4815111] CHIP: [EM] Rxd Ack; Removing MessageCounter:5009094 from Retrans Table on exchange 61488i
[1652983963053] [77495:4815111] CHIP: [EM] Removed CHIP MessageCounter:5009094 from RetransTable on exchange 61488i
[1652983963053] [77495:4815111] CHIP: [DMG] ReportDataMessage =
[1652983963053] [77495:4815111] CHIP: [DMG] {
[1652983963053] [77495:4815111] CHIP: [DMG] 	AttributeReportIBs =
[1652983963053] [77495:4815111] CHIP: [DMG] 	[
[1652983963053] [77495:4815111] CHIP: [DMG] 		AttributeReportIB =
[1652983963053] [77495:4815111] CHIP: [DMG] 		{
[1652983963053] [77495:4815111] CHIP: [DMG] 			AttributeDataIB =
[1652983963053] [77495:4815111] CHIP: [DMG] 			{
[1652983963053] [77495:4815111] CHIP: [DMG] 				DataVersion = 0x584ebb85,
[1652983963053] [77495:4815111] CHIP: [DMG] 				AttributePathIB =
[1652983963053] [77495:4815111] CHIP: [DMG] 				{
[1652983963053] [77495:4815111] CHIP: [DMG] 					Endpoint = 0x0,
[1652983963053] [77495:4815111] CHIP: [DMG] 					Cluster = 0x3e,
[1652983963053] [77495:4815111] CHIP: [DMG] 					Attribute = 0x0000_0001,
[1652983963053] [77495:4815111] CHIP: [DMG] 				}
[1652983963053] [77495:4815111] CHIP: [DMG] 					
[1652983963053] [77495:4815111] CHIP: [DMG] 					Data = [
[1652983963054] [77495:4815111] CHIP: [DMG] 						
[1652983963054] [77495:4815111] CHIP: [DMG] 					],
[1652983963054] [77495:4815111] CHIP: [DMG] 			},
[1652983963054] [77495:4815111] CHIP: [DMG] 			
[1652983963054] [77495:4815111] CHIP: [DMG] 		},
[1652983963054] [77495:4815111] CHIP: [DMG] 		
[1652983963054] [77495:4815111] CHIP: [DMG] 		AttributeReportIB =
[1652983963054] [77495:4815111] CHIP: [DMG] 		{
[1652983963054] [77495:4815111] CHIP: [DMG] 			AttributeDataIB =
[1652983963054] [77495:4815111] CHIP: [DMG] 			{
[1652983963054] [77495:4815111] CHIP: [DMG] 				DataVersion = 0x584ebb85,
[1652983963054] [77495:4815111] CHIP: [DMG] 				AttributePathIB =
[1652983963054] [77495:4815111] CHIP: [DMG] 				{
[1652983963054] [77495:4815111] CHIP: [DMG] 					Endpoint = 0x0,
[1652983963054] [77495:4815111] CHIP: [DMG] 					Cluster = 0x3e,
[1652983963054] [77495:4815111] CHIP: [DMG] 					Attribute = 0x0000_0001,
[1652983963054] [77495:4815111] CHIP: [DMG] 					ListIndex = Null,
[1652983963054] [77495:4815111] CHIP: [DMG] 				}
[1652983963054] [77495:4815111] CHIP: [DMG] 					
[1652983963054] [77495:4815111] CHIP: [DMG] 					Data = 
[1652983963054] [77495:4815111] CHIP: [DMG] 					{
[1652983963054] [77495:4815111] CHIP: [DMG] 						0x1 = [
[1652983963054] [77495:4815111] CHIP: [DMG] 							0x4, 0xae, 0x53, 0xf6, 0x3c, 0x77, 0x5e, 0x2c, 0xf6, 0x1f, 0xf1, 0x6c, 0x89, 0x13, 0x86, 0x79, 0x49, 0x87, 0x7d, 0x78, 0x38, 0x55, 0x8a, 0x41, 0xc8, 0x69, 0xb6, 0xa0, 0xc9, 0xeb, 0x12, 0x1a, 0x19, 0x72, 0
[1652983963054] [77495:4815111] CHIP: [DMG] 						]
[1652983963054] [77495:4815111] CHIP: [DMG] 						0x2 = 65521, 
[1652983963054] [77495:4815111] CHIP: [DMG] 						0x3 = 1, 
[1652983963054] [77495:4815111] CHIP: [DMG] 						0x4 = 1, 
[1652983963054] [77495:4815111] CHIP: [DMG] 						0x5 = "", 
[1652983963054] [77495:4815111] CHIP: [DMG] 						0xfe = 1, 
[1652983963054] [77495:4815111] CHIP: [DMG] 					},
[1652983963054] [77495:4815111] CHIP: [DMG] 			},
[1652983963054] [77495:4815111] CHIP: [DMG] 			
[1652983963054] [77495:4815111] CHIP: [DMG] 		},
[1652983963054] [77495:4815111] CHIP: [DMG] 		
[1652983963054] [77495:4815111] CHIP: [DMG] 		AttributeReportIB =
[1652983963054] [77495:4815111] CHIP: [DMG] 		{
[1652983963054] [77495:4815111] CHIP: [DMG] 			AttributeDataIB =
[1652983963054] [77495:4815111] CHIP: [DMG] 			{
[1652983963054] [77495:4815111] CHIP: [DMG] 				DataVersion = 0x584ebb85,
[1652983963054] [77495:4815111] CHIP: [DMG] 				AttributePathIB =
[1652983963054] [77495:4815111] CHIP: [DMG] 				{
[1652983963054] [77495:4815111] CHIP: [DMG] 					Endpoint = 0x0,
[... 122 lines cut ...]
[1652984383336] [77589:4819752] CHIP: [DMG] }
[1652984383337] [77589:4819752] CHIP: [TOO] Endpoint: 0 Cluster: 0x0000_003E Attribute 0x0000_0001 DataVersion: 3117813818
[1652984383337] [77589:4819752] CHIP: [TOO]   Fabrics: 2 entries
[1652984383337] [77589:4819752] CHIP: [TOO]     [1]: {
[1652984383337] [77589:4819752] CHIP: [TOO]       RootPublicKey: 04A2088FA46BACAA4B2E24AE58FDE5E82119EEF8450F058D4E196E246506EED75A6F5F16CD355C4F0272A465272F85201699690F67AC27F3C6B9BC153837F58EAB
[1652984383337] [77589:4819752] CHIP: [TOO]       VendorId: 65521
[1652984383337] [77589:4819752] CHIP: [TOO]       FabricId: 1
[1652984383337] [77589:4819752] CHIP: [TOO]       NodeId: 1
[1652984383337] [77589:4819752] CHIP: [TOO]       Label: 
[1652984383337] [77589:4819752] CHIP: [TOO]       FabricIndex: 1
[1652984383337] [77589:4819752] CHIP: [TOO]      }
[1652984383337] [77589:4819752] CHIP: [TOO]     [2]: {
[1652984383337] [77589:4819752] CHIP: [TOO]       RootPublicKey: 04A2088FA46BACAA4B2E24AE58FDE5E82119EEF8450F058D4E196E246506EED75A6F5F16CD355C4F0272A465272F85201699690F67AC27F3C6B9BC153837F58EAB
[1652984383337] [77589:4819752] CHIP: [TOO]       VendorId: 65521
[1652984383337] [77589:4819752] CHIP: [TOO]       FabricId: 2
[1652984383337] [77589:4819752] CHIP: [TOO]       NodeId: 1
[1652984383337] [77589:4819752] CHIP: [TOO]       Label: 
[1652984383337] [77589:4819752] CHIP: [TOO]       FabricIndex: 2
[1652984383337] [77589:4819752] CHIP: [TOO]      }
[1652984383338] [77589:4819752] CHIP: [EM] Sending Standalone Ack for MessageCounter:5611497 on exchange 20666i
[1652984383338] [77589:4819752] CHIP: [IN] Prepared secure message 0x70000963c560 to 0x0000000000000001 (1)  of type 0x10 and protocolId (0, 0) on exchange 20666i with MessageCounter:10762022.
[1652984383338] [77589:4819752] CHIP: [IN] Sending encrypted msg 0x70000963c560 with MessageCounter:10762022 to 0x0000000000000001 (1) at monotonic time: 000000001F4F4361 msec
[1652984383338] [77589:4819752] CHIP: [EM] Flushed pending ack for MessageCounter:5611497 on exchange 20666i
[1652984383338] [77589:4819748] CHIP: [CTL] Shutting down the commissioner
[1652984383338] [77589:4819748] CHIP: [CTL] Shutting down the controller

bzbarsky-apple, 2022-05-24:
See also https://github.com/project-chip/connectedhomeip/issues/16148

andy31415, 2022-05-26:
@Garik098 - is this chip-tool-darwin specific or does it affect all chip-tools?

andy31415, 2022-05-26:
@cjandhyala you added cert-blocker however this seems like maybe platform specific (darwin)? Could you confirm if this is actually a cert blocker?

Garik098, 2022-05-26:
> @Garik098 - is this chip-tool-darwin specific or does it affect all chip-tools?

This is chip tool Darwin specific and the issue is not being seen in chip tool.

PR #18909 (2022-06-02): Add chip tool Darwin description to classes.
#### Problem
When logging a structure class, it does not log any of the attributes. This makes it difficult to validate responses using darwin-framework-tool.
- Fixes #18634 
- Fixes #16148 

#### Change overview
- Updated ZAP files that create the structure classes to include a description function. 
- Update darwin-framework-tool to iterated and log 

#### Testing
- Validated that the darwin-framework-tool display values in classes, even within a list.
closed 2022-06-02 by woody-apple