project-chip/connectedhomeip #18644: [TE9] chip-tool segmentation fault when running TC-PS-2.2

ifyall, 2022-05-19:
#### Problem
When running TC-PS-2.2 in TE9, the chip-tool crashed with a segmentation fault when executing the command 'powersource subscribe-event-by-id'

> ubuntu@ubuntu:~$ chip-tool powersource subscribe-event-by-id 0 120 3600 12344321 1
> [1652385557.647020][34542:34542] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
> [1652385557.647561][34542:34542] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
> [1652385557.647722][34542:34542] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
> [1652385557.647830][34542:34542] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
> [1652385557.648183][34542:34542] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-iDi3au)
> [1652385557.648865][34542:34542] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1652385557.648925][34542:34542] CHIP:DL: NVS set: chip-counters/reboot-count = 114 (0x72)
> [1652385557.649768][34542:34542] CHIP:DL: Got Ethernet interface: eth0
> [1652385557.650471][34542:34542] CHIP:DL: Found the primary Ethernet interface:eth0
> [1652385557.651172][34542:34542] CHIP:DL: Got WiFi interface: wlan0
> [1652385557.651256][34542:34542] CHIP:DL: Failed to reset WiFi statistic counts
> [1652385557.651344][34542:34542] CHIP:IN: UDP::Init bind&listen port=0
> [1652385557.651488][34542:34542] CHIP:IN: UDP::Init bound to port=45219
> [1652385557.651519][34542:34542] CHIP:IN: BLEBase::Init - setting/overriding transport
> [1652385557.651543][34542:34542] CHIP:IN: TransportMgr initialized
> [1652385557.651605][34542:34542] CHIP:DIS: Init fabric pairing table with server storage
> [1652385557.651936][34542:34542] CHIP:IN: Loading from storage for fabric index 0x1
> [1652385557.655385][34542:34542] CHIP:IN: Loading from storage for fabric index 0x2
> [1652385557.656658][34542:34542] CHIP:IN: Loading from storage for fabric index 0x3
> [1652385557.657874][34542:34542] CHIP:DIS: Add fabric pairing table delegate
> [1652385557.664710][34542:34542] CHIP:ZCL: Using ZAP configuration...
> [1652385557.668967][34542:34542] CHIP:DL: Avahi client registered
> [1652385557.670211][34542:34542] CHIP:CTL: System State Initialized...
> [1652385557.670299][34542:34542] CHIP:CTL: Setting attestation nonce to random value
> [1652385557.670388][34542:34542] CHIP:CTL: Setting CSR nonce to random value
> [1652385557.670515][34542:34542] CHIP:CTL: Setting attestation nonce to random value
> [1652385557.670565][34542:34542] CHIP:CTL: Setting CSR nonce to random value
> [1652385557.671490][34542:34542] CHIP:CTL: Generating NOC
> [1652385557.672461][34542:34542] CHIP:DIS: Verifying the received credentials
> [1652385557.674905][34542:34542] CHIP:DIS: Added new fabric at index: 0x1, Initialized: 1
> [1652385557.674956][34542:34542] CHIP:DIS: Assigned compressed fabric ID: 0x01E738535DD7208F, node ID: 0x000000000001B669
> [1652385557.674989][34542:34542] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x01E738535DD7208F
> [1652385557.675098][34542:34542] CHIP:CTL: Setting attestation nonce to random value
> [1652385557.675180][34542:34542] CHIP:CTL: Setting CSR nonce to random value
> [1652385557.676304][34542:34542] CHIP:CTL: Generating NOC
> [1652385557.677522][34542:34542] CHIP:DIS: Verifying the received credentials
> [1652385557.680101][34542:34542] CHIP:DIS: Added new fabric at index: 0x2, Initialized: 1
> [1652385557.680155][34542:34542] CHIP:DIS: Assigned compressed fabric ID: 0x4D808C102B78AC41, node ID: 0x000000000001B669
> [1652385557.680190][34542:34542] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x4D808C102B78AC41
> [1652385557.680293][34542:34542] CHIP:CTL: Setting attestation nonce to random value
> [1652385557.680375][34542:34542] CHIP:CTL: Setting CSR nonce to random value
> [1652385557.681503][34542:34542] CHIP:CTL: Generating NOC
> [1652385557.682917][34542:34542] CHIP:DIS: Verifying the received credentials
> [1652385557.684699][34542:34542] CHIP:DIS: Added new fabric at index: 0x3, Initialized: 1
> [1652385557.684736][34542:34542] CHIP:DIS: Assigned compressed fabric ID: 0xA5A256CC44153F13, node ID: 0x000000000001B669
> [1652385557.684760][34542:34542] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0xA5A256CC44153F13
> [1652385557.764727][34542:34547] CHIP:DL: CHIP task running
> [1652385557.764958][34542:34547] CHIP:DL: HandlePlatformSpecificBLEEvent 32787
> [1652385557.764992][34542:34547] CHIP:TOO: Sending command to node 0xbc5c01
> [1652385557.765027][34542:34547] CHIP:CSM: FindOrEstablishSession: PeerId = 01E738535DD7208F:0000000000BC5C01
> [1652385557.765054][34542:34547] CHIP:CSM: FindOrEstablishSession: No existing OperationalDeviceProxy instance found
> [1652385557.765097][34542:34547] CHIP:DIS: Resolving 01E738535DD7208F:0000000000BC5C01 ...
> [1652385557.766989][34542:34547] CHIP:DL: Avahi resolve found
> [1652385557.767071][34542:34547] CHIP:DIS: Node ID resolved for 01E738535DD7208F:0000000000BC5C01
> [1652385557.767104][34542:34547] CHIP:DIS:     Addr 0: [2601:1c0:4701:53b0:c6ac:59ff:fe9c:6454]:5540
> [1652385557.768006][34542:34547] CHIP:DIS: Lookup clearing interface for non LL address
> [1652385557.768043][34542:34547] CHIP:DIS: UDP:[2601:1c0:4701:53b0:c6ac:59ff:fe9c:6454%eth0]:5540: new best score: 7
> [1652385557.768067][34542:34547] CHIP:DIS: Checking node lookup status after 3 ms
> [1652385557.768089][34542:34547] CHIP:DIS: Keeping DNSSD lookup active
> [1652385557.965480][34542:34547] CHIP:DIS: Checking node lookup status after 201 ms
> [1652385557.965599][34542:34547] CHIP:CTL: Updating device address to UDP:[2601:1c0:4701:53b0:c6ac:59ff:fe9c:6454]:5540 while in state 1
> [1652385557.965656][34542:34547] CHIP:CTL: OperationalDeviceProxy[01E738535DD7208F:0000000000BC5C01]: State change 1 --> 2
> [1652385557.966085][34542:34547] CHIP:SC: Initiating session on local FabricIndex 1 from 0x000000000001B669 -> 0x0000000000BC5C01
> [1652385557.968220][34542:34547] CHIP:SC: Including MRP parameters
> [1652385557.968989][34542:34547] CHIP:IN: Prepared unauthenticated message 0xaaaad7bced08 to 0x0000000000000000 (0)  of type 0x30 and protocolId (0, 0) on exchange 23458i with MessageCounter:2334500426.
> [1652385557.969095][34542:34547] CHIP:IN: Sending unauthenticated msg 0xaaaad7bced08 with MessageCounter:2334500426 to 0x0000000000000000 at monotonic time: 000000000447263E msec
[... 61 lines cut ...]
> [1652385587.765472][34542:34542] CHIP:IN: Expiring all connections for fabric 1!!
> [1652385587.765560][34542:34542] CHIP:CTL: Shutting down the commissioner
> [1652385587.765613][34542:34542] CHIP:CTL: Shutting down the controller
> [1652385587.765658][34542:34542] CHIP:IN: Expiring all connections for fabric 2!!
> [1652385587.765727][34542:34542] CHIP:CTL: Shutting down the commissioner
> [1652385587.765775][34542:34542] CHIP:CTL: Shutting down the controller
> [1652385587.765818][34542:34542] CHIP:IN: Expiring all connections for fabric 3!!
> [1652385587.765878][34542:34542] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
> [1652385587.767634][34542:34542] CHIP:DMG: IM WH moving to [Uninitialized]
> [1652385587.767703][34542:34542] CHIP:DMG: IM WH moving to [Uninitialized]
> [1652385587.767746][34542:34542] CHIP:DMG: IM WH moving to [Uninitialized]
> [1652385587.767788][34542:34542] CHIP:DMG: IM WH moving to [Uninitialized]
> [1652385587.767833][34542:34542] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
> [1652385587.767959][34542:34542] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
> [1652385587.768403][34542:34542] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-h0uiVs)
> [1652385587.769574][34542:34542] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
> [1652385587.769697][34542:34542] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
> [1652385587.769752][34542:34542] CHIP:DL: Inet Layer shutdown
> [1652385587.769797][34542:34542] CHIP:DL: BLE shutdown
> [1652385587.769843][34542:34542] CHIP:DL: System Layer shutdown
> [1652385587.770065][34542:34542] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:392: CHIP Error 0x00000032: Timeout
> Segmentation fault (core dumped)

 #### Proposed Solution
Need investigation of the root cause of the segmentation fault.

isiu-apple, 2022-05-19:
For event subscriptions, try running chiptool in interactive mode:

```
chip-tool interactive start
chip-tool powersource subscribe-event-by-id 0 120 3600 12344321 1
```

andy31415, 2022-05-20:
Segfault is likely unclear shutdown due to timeout

bzbarsky-apple, 2022-05-21:
@ifyall Is this reproducible on ToT?  I just tried running that command and I am not seeing a segfault...

bzbarsky-apple, 2022-05-25:
@cjandhyala You added the "cert blocker" label, but is this reproducible?

andy31415, 2022-05-25:
assigning this to @cjandhyala - need ToT reproduction steps (since Boris tried and it did not reproduce) or closing.

@ifyall could you also add info regarding setup:
   - what platforms was the command run on
   - how often does it reproduce
   - if you are able to re-run it quickly enough, can you test on ToT on your environment?
   
 We also now have a doc: https://github.com/project-chip/connectedhomeip/blob/master/docs/BUG_REPORT.md . It generally asks for platform info and also asks for logs to be provided as attachments (and only curated subsets being inline) to avoid making the issue description too large.

billwatersiii, 2022-05-27:
I do not see the Segmentation fault on the ToT.

@ifyall, I guess we can close this with a "can not reproduce" or "it just went away".

closed 2022-05-27 by ifyall