project-chip/connectedhomeip #18646: Test case to Simulate disconnecting and reconnecting to a Thread network on TH fails with : Error 0x0000002F

phonnakasturi-apple, 2022-05-20:
Test case to Simulate disconnecting and reconnecting to a Thread network on TH fails:

**Python command:**
sudo ./chip-tool-darwin  threadnetworkdiagnostics  read-event connection-status 30 0

**Error**
[1653006314770] [95607:996751] CHIP: [TOO] Run command failure: ../../third_party/connectedhomeip/examples/chip-tool/commands/common/Commands.cpp:113: Error 0x0000002F

Attaching the screenshot;

[image]

bzbarsky-apple, 2022-05-25:
chip-tool-darwin does not support reading events, because the Darwin framework does not.

bzbarsky-apple, 2022-05-25:
Duplicate of #18185

closed 2022-05-25 by bzbarsky-apple