project-chip/connectedhomeip #18647: [TC-DD-3.20] All cluaster app can't scan after remove fabric.

joonhaengHeo, 2022-05-20:
#### Problem
All cluaster app can't scan after remove fabric.
And, "sudo ./chip-tool discover commissionables" command also can't scan.

Remove Fabric Log.

[1653006728.794110][9253:9253] CHIP:EM: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:3683182 on exchange 57427r
[1653006728.794242][9253:9253] CHIP:EM: Handling via exchange: 57427r, Delegate: 0xaaaae8a7b0a0
[1653006728.794393][9253:9253] CHIP:DMG: InvokeRequestMessage =
[1653006728.794458][9253:9253] CHIP:DMG: {
[1653006728.794514][9253:9253] CHIP:DMG:        suppressResponse = false,
[1653006728.794579][9253:9253] CHIP:DMG:        timedRequest = false,
[1653006728.794639][9253:9253] CHIP:DMG:        InvokeRequests =
[1653006728.794753][9253:9253] CHIP:DMG:        [
[1653006728.794813][9253:9253] CHIP:DMG:                CommandDataIB =
[1653006728.794918][9253:9253] CHIP:DMG:                {
[1653006728.794982][9253:9253] CHIP:DMG:                        CommandPathIB =
[1653006728.795075][9253:9253] CHIP:DMG:                        {
[1653006728.795162][9253:9253] CHIP:DMG:                                EndpointId = 0x0,
[1653006728.795243][9253:9253] CHIP:DMG:                                ClusterId = 0x3e,
[1653006728.795345][9253:9253] CHIP:DMG:                                CommandId = 0xa,
[1653006728.795429][9253:9253] CHIP:DMG:                        },
[1653006728.795537][9253:9253] CHIP:DMG:
[1653006728.795606][9253:9253] CHIP:DMG:                        CommandData =
[1653006728.795706][9253:9253] CHIP:DMG:                        {
[1653006728.795793][9253:9253] CHIP:DMG:                                0x0 = 1,
[1653006728.795937][9253:9253] CHIP:DMG:                        },
[1653006728.796010][9253:9253] CHIP:DMG:                },
[1653006728.796087][9253:9253] CHIP:DMG:
[1653006728.796166][9253:9253] CHIP:DMG:        ],
[1653006728.796238][9253:9253] CHIP:DMG:
[1653006728.796318][9253:9253] CHIP:DMG:        InteractionModelRevision = 1
[1653006728.796375][9253:9253] CHIP:DMG: },
[1653006728.796535][9253:9253] CHIP:DMG: AccessControl: checking f=1 a=c s=0x000000000001B669 t= c=0x0000_003E e=0 p=a
[1653006728.796636][9253:9253] CHIP:DMG: AccessControl: allowed
[1653006728.796704][9253:9253] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_003E Command=0x0000_000A
[1653006728.796774][9253:9253] CHIP:ZCL: OpCreds: RemoveFabric

 #### Proposed Solution
Commissionee can discover after remove fabric (Commissioned device is 0.)

andy31415, 2022-05-20:
@joonhaengHeo could you provide details on test setup?

- what is the platform being tested: specifically the DUT (like linux/mac/nrf/esp32/efr32/qpg/other)
- what are the commands that are being executed to reproduce

I imagine it is some form of commission remove fabric and rescan. Please list all commands so we can reproduce.

bzbarsky-apple, 2022-05-22:
Removing a fabric does not automatically make a device commissionable again.

closed 2022-05-22 by bzbarsky-apple