project-chip/connectedhomeip #18663: [TC-MC-3.7]Application Launcher Cluster--Chip-tool needs to provide a way for TH to sends a LaunchApp command to DUT to launch an app and bring an application to the foreground.

Rajashreekalmane, 2022-05-20:
#### Problem
[TC-MC-3.7]Application Launcher Cluster--Chip-tool needs to provide a way for TH to sends a LaunchApp command to DUT to launch an app and bring an application to the foreground.

**Test-Plan screenshot**
[image]

andy31415, 2022-05-20:
@chrisdecenzo - would this fall under your development as this involves application launcher cluster?

chrisdecenzo, 2022-05-20:
You can use the launch-app command. For example, the following works to launch an app in the tv-app example:

chip-tool applicationlauncher launch-app '{"catalogVendorId": 123, "applicationId": "exampleid"}' 1234 1

I added this instruction to the notes section of this test case in the test plan doc.

closed 2022-05-20 by chrisdecenzo