project-chip/connectedhomeip #18666: [TC-IDM-3.1]Chip-tool support needed to  send a ReadRequest message from TH to DUT to read any attribute on two clusters.

Rajashreekalmane, 2022-05-20:
**#### Problem**
As per Test-Plan, test-step 13 says "DUT sends a ReadRequest message to the TH to read any attribute on two clusters", 
Chip-tool support needed to  send a ReadRequest message from TH to DUT to read any attribute on two clusters.

**Test-Plan screenshot:-**
[image]

**Spec Reference:-**
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#ref_WriteRequestAction

andy31415, 2022-05-20:
@Rajashreekalmane - I am a bit unclear here, trying to determine if the issue is `any` being not strictly defined or if chip-tool does not have the ability to read or write two attributes at the same time or something else.

Please describe the issue a bit more. Generally I would expect replacing `any` with a concrete example maybe of some attribute that is always defined and can be changed (like labels maybe?).

Rajashreekalmane, 2022-05-20:
@andy31415 chip-tool does not have the ability read or write any attributes from different clusters at the same time,  As my understanding here any attributes means we need to send command which should contain different attributes from different clusters.

bzbarsky-apple, 2022-05-22:
> chip-tool does not have the ability read or write any attributes from different clusters at the same time

Sure it does:
```
chip-tool any read-by-id "6,8" "0,17" ${NOD_ID} "1,1"
```
will read the "OnOff" and "OnLevel" attributes from the "On/Off" and "Level Control" clusters respectively on endpoint 1.  Resulting output is:
```
[1653195653340] [45076:53049833] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0008 Attribute 0x0000_0011 DataVersion: 1642453931
[1653195653341] [45076:53049833] CHIP: [TOO]   on level: null
[1653195653341] [45076:53049833] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 2463002072
[1653195653341] [45076:53049833] CHIP: [TOO]   OnOff: FALSE
```

Adjust the IDs as needed.

bzbarsky-apple, 2022-05-22:
Or you can do:
```
chip-tool any read-by-id "6,8" "0,17" ${NODE_ID} "2,1"
```
to have different endpoints too:
```
[1653195823045] [45084:53051019] CHIP: [TOO] Endpoint: 1 Cluster: 0x0000_0008 Attribute 0x0000_0011 DataVersion: 1642453931
[1653195823045] [45084:53051019] CHIP: [TOO]   on level: null
[1653195823045] [45084:53051019] CHIP: [TOO] Endpoint: 2 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 2639132549
[1653195823045] [45084:53051019] CHIP: [TOO]   OnOff: FALSE
```

closed 2022-05-22 by bzbarsky-apple