project-chip/connectedhomeip #18681: [Door-Lock-Server] Calls to emberAfPluginDoorLockOnUnhandledAttributeChange need added attributePath parameter

jrhees-cae, 2022-05-20:
#### Problem
The `emberAfPluginDoorLockOnUnhandledAttributeChange` function needs the attributePath parameter added to the parameter list.

 #### Proposed Solution
change function declaration from:
`emberAfPluginDoorLockOnUnhandledAttributeChange(chip::EndpointId EndpointId, EmberAfAttributeType attrType, uint16_t attrSize, uint8_t * attrValue)`
to:
`emberAfPluginDoorLockOnUnhandledAttributeChange(chip::EndpointId EndpointId, const chip::app::ConcreteAttributePath & attributePath, EmberAfAttributeType attrType, uint16_t attrSize, uint8_t * attrValue)`

mrjerryjohns, 2022-05-23:
@Morozov-5F would you be able to help take this on?

bzbarsky-apple, 2022-05-25:
@jrhees-cae what are the observable consequences of this function having the wrong signature?

jrhees-cae, 2022-05-25:
Without the attribute path being included as a parameter, the function is
completely useless to an app which wishes to override the weak link. The
app gets the call, but had no way of knowing which attribute was being
changed.

Jon

bzbarsky-apple, 2022-07-02:
Sounds like lock apps will need this....

woody-apple, 2022-07-05:
SDK Review: Seems like this could help Lock's for SVE, not required, but given where we are right now, we can take this.

woody-apple, 2022-07-05:
SDK Review: @Morozov-5F are you able to take a look at this?

PR #20338 (2022-07-06): Add attributePath parameter to Door Lock Server's unhandledAttributeC…
Addd attributePath parameter to Door Lock Server's emberAfPluginDoorLockOnUnhandledAttributeChanged API

#### Problem
API function needs attributePath parameter to allow application to determine which attribute is being changed.

#### Change overview
Add attributePath parameter API call

#### Testing
No existing code is affected (e.g., API is not currently used by any examples).

Fixes #18681
closed 2022-07-06 by woody-apple