project-chip/connectedhomeip #18686: [Java] No way to read attributes without fabric-filtering

g-coppock, 2022-05-20:
#### Problem

In the generated cluster code, for example in `OperationalCredentialsClient-ReadImpl.cpp`, the `ReadAttribute` method has a default boolean parameter for `aIsFabricFiltered` which defaults to `true`.

As this generated code does not provide any mechanism by which to set this attribute, all reads are fabric-filtered.

 #### Proposed Solution

Suggested solutions:

1. Generate overloaded methods for each `readX` method which accept a boolean for `fabricFiltered`. Probably the simplest option, but the downside here is a lot of extra code, and a growing parameter list.
2. Update the generated methods to require a boolean argument for `fabricFiltered`. Also simple, but is a breaking change for Java clients.
3. Add a Java-layer `ReadParams` class that contains any arguments that we want to surface (timeout, fabricFiltered) to mirror `ReadPrepareParams`, and have the generated code accept this as an argument. This would also be a breaking change, but would simplify later addition of any other arguments for reading attributes without breaking compatibility. Downsides here are that `ReadPrepareParams` has overlap between read/subscribe, so this would add parameters to the read methods that may be meaningless (e.g. `mMinIntervalFloorSeconds` for a read).

yunhanw-google, 2022-06-06:
Plan to have Java class ReadPrepareParams

andy31415, 2022-06-14:
#19437 adds options for fabric filtering

PR #22575 (2022-09-16): [Android/JAVA]add IM isFabricFiltered and keepSubscriptions flag for read/subscribe API
#### Issue Being Resolved
* Fixes #18686 

#### Change overview
Add IM isFabricFiltered and keepSubscriptions flag for read/subscribe API, fix across jni, java, kotlin, android chip-tool.
Disable auto-resubscribe since controller jni is using std::vector, we cannot move the memory to readclient.
closed 2022-09-16 by andy31415