project-chip/connectedhomeip #18727: No support commissionee device to delete fabric

nicelyjust, 2022-05-23:
#### Problem
On some devices with a screen, if user want to delete a fabric that has commissioned,how to do it?

 #### Proposed Solution
I think this is a very common feature for users, I hope it can be supported

mrjerryjohns, 2022-05-23:
Please check out the `RemoveFabric` command on the `OperationalCredentialsCluster`. That requires a fabric-index of the fabric to be deleted to be passed in. The fabric index that corresponds to the right fabric can be retrieved by reading the `Fabrics` list and upon finding the right entry, to retrieve the 'fabric-index' field from within that entry.

mrjerryjohns, 2022-05-23:
Please close if the above comment resolves your question.

nicelyjust, 2022-05-24:
Thank you for the response, @mrjerryjohns 
OperationalCredentialsCluster is responsible for commissioner ->commissionee, I mean that adding fabric deletion methods to allow the user to delete a fabric on device side,not through commissioner

bzbarsky-apple, 2022-05-25:
Generally speaking, the intent is that only a valid administrator of the device will be able to remove fabrics, right?

nicelyjust, 2022-05-25:
yes, For example:
[image]

nicelyjust, 2022-05-25:
Yes, For example I want to manager fabrics on TV, GUI like this
[image]

closed 2022-12-03 by stale[bot]