project-chip/connectedhomeip #18737: Nordic subtype registration didn't match base type service

kpark-apple, 2022-05-23:
#### Problem
Nordic pairing failed with an OnSRPClientNotification: Operation refused for security reasons.
In this particular case, on the SRP server side, it was reported that _V65521._sub._matterc._udp.default.service.arpa record had 6A49452171E45256._matterc._udp.default.service.arpa base type which seemed to have been removed.

It appears that there might have been sync issue in removing old services and adding new services in Nordic platform. I think that the subtype should have been registered with new service 26820495907622CE._matterc._udp instead.

```
I: 55506 [DIS]Updating services using commissioning mode 2                      
D: 55512 [DL]Using Thread extended MAC for hostname.                            
I: 55516 [DIS]Advertise operational node 374DA47E60686690-00000000B53BC742      
I: 55523 [DL]advertising srp service: 374DA47E60686690-00000000B53BC742._matterp
D: 55531 [DL]Using Thread extended MAC for hostname.                            
I: 55649 [DIS]Advertise commission parameter vendorID=65521 productID=32773 dis7
I: 55658 [DL]advertising srp service: 26820495907622CE._matterc._udp            
I: 55664 [DL]removing srp service: 6A49452171E45256._matterc._udp 
```

 #### Proposed Solution
N/A

[Nordic_pairing_attempt_0520173500EST.TXT](https://github.com/project-chip/connectedhomeip/files/8756963/Nordic_pairing_attempt_0520173500EST.TXT)

bzbarsky-apple, 2022-05-25:
@Damian-Nordic could you take a look please?

bzbarsky-apple, 2022-05-25:
@kpark-apple Is this the same issue as #18507?

kpark-apple, 2022-05-25:
> @kpark-apple Is this the same issue as #18507?

No, it's a different issue. In this issue, border router refused while in #18507, border router accepted and successfully finished all updates while the Nordic log still showed refused.

kpark-apple, 2022-05-25:
> @kpark-apple Do you have any logs from the border router? Are you using OTBR? Which version? I'm asking because, we haven't seen this error for months and we do a lot of testing against OTBR, so this might be some border router issue as well.

@Damian-Nordic Border router isn't OTBR. But the error is from srp-mdns-proxy in our border router. We have repeated log of the following:

```
2022-05-20 17:38:29.445576-0400 0xee1      Error       0x0                  119    0    srp-mdns-proxy: srp_evaluate: service subtype _V65521._sub._matterc._udp.default.service.arpa. for 6A49452171E45256._matterc._udp.default.service.arpa. doesn't match base type service 26820495907622CE._matterc._udp.default.service.arpa.
```

bzbarsky-apple, 2022-05-25:
@Abhayakara

Abhayakara, 2022-05-25:
What's going on here is that we didn't anticipate a need to publish a new service instance of the same service type on the same host, so the error-checking code is not accounting for the case where this is what's happening, and is instead seeing it as a mistake.

kpark-apple, 2022-05-25:
Closing this for Apple border router has to allow this sequence of updates.

closed 2022-05-25 by kpark-apple

Abhayakara, 2022-05-26:
Can I get a packet trace for this?