project-chip/connectedhomeip #18745: MdnsAvahi: Init causes chip-tool to exit

bluebin14, 2022-05-24:
#### Problem
Seen with TE9 image in several configurations, chip-tool was exiting because MdnsAvahi::Init was called twice, making it impossible to use.

 #### Proposed Solution
Do not fail when instance is reinited. See also https://github.com/project-chip/connectedhomeip/pull/18670

bzbarsky-apple, 2022-05-24:
The key part here is that Init() on the platform mdns stuff needs to be a no-op after the first call.

Right now it might not be handling that right because parts of Init() are async, so it can end up setting the "I am initialized" boolean much later.

bzbarsky-apple, 2022-07-26:
The crash is gone, but the underlying "API docs say it's a no-op, but it's not a no-op" behavior is still there...