project-chip/connectedhomeip #18750: FeatureMap returns 0x86 (UNSUPPORTED_ATTRIBUTE) in multiple clusters.

Rajashreekalmane, 2022-05-24:
**#### Problem**
Though the feature map attribute is implemented in below mentioned clusters, still it throws an error code as 0x86(unsupported attribute).
 Thermostatuserinterfaceconfiguration cluster
Temperaturemeasurement cluster
 Illuminancemeasurement cluster

**expected behavior:-**FeatureMap bitmap value 0
**actual behavior:-**0x86(unsupported attribute

 ***** steps to reproduce****
 ./chip-tool   illuminancemeasurement  read feature-map 1 1
./chip-tool   thermostatuserinterfaceconfiguration   read feature-map 1 1
./chip-tool  temperaturemeasurement   read feature-map 1 1

****system configuration:-****
Chip-Tool: RPI(chip-tool)
DUT: RPI(all-cluster-app)
Platform : IP
Commit-ID: https://github.com/project-chip/connectedhomeip/commit/7afd35229da424a139642e4ebbe68bb4d71554cf
Platform:-Linux platform

**Spec reference:-**
[image]

**Test-Plan reference:-**
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/thermostatuserconfiguration.adoc#tc-tsuic-1-1-global-attributes-with-server-as-dut
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/temperaturemeasurement.adoc#tc-tm-1-1-global-attributes-with-server-as-dut
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/illuminance_measurement.adoc#tc-ill-1-1-global-attributes-with-server-as-dut

**PFA log:-**
[feature map.txt](https://github.com/project-chip/connectedhomeip/files/8762317/feature.map.txt)

somu1710, 2022-05-24:
Facing the same issue as error code as 0x86(unsupported attribute) in below mentioned test cases:
1.Pressure measurement cluster
2.Occupancy sensing cluster
3.Relative humidity
4.Power Source Configuration
5.Descriptor

bzbarsky-apple, 2022-05-25:
> Commit-ID: https://github.com/project-chip/connectedhomeip/commit/7afd35229da424a139642e4ebbe68bb4d71554cf

Please don't bother testing FeatureMap (or at leats reporting issues about it) on anything before 2db812e8ef2f7fe3edfe04da4b526b3e977a1a43.

I just tried all the clusters listed in https://github.com/project-chip/connectedhomeip/issues/18750#issuecomment-1136066609 in all-clusters app on tip  (fda7854efb39fbfac9dd1063cc428c17b9676970) and they all work correctly.

closed 2022-05-25 by bzbarsky-apple