project-chip/connectedhomeip #18761: Re-subscription after restarting device

dtodor, 2022-05-24:
We have a Linux controller, which subscribes to the `OnOff::OnOff` attribute of a EFR32 lighting-app device. The subscription and attribute notifications work as expected. However, if we power-cycle the EFR32 device, the automatic re-subscription fails. Looking at the logs on both the Linux and EFR32 sides, it seems that the CASE session is not re-established:

Linux Controller
```
[1653405607.963477][8491:8492] CHIP:IN: Prepared secure message 0x56440cc4 to 0x0000000000000457 (1)  of type 0x3 and protocolId (0, 1) on exchange 52699i with MessageCounter:13517467.
[1653405607.963641][8491:8492] CHIP:IN: Sending encrypted msg 0x56440cc4 with MessageCounter:13517467 to 0x0000000000000457 (1) at monotonic time: 0000000000AC2001 msec
[1653405607.964250][8491:8492] CHIP:DMG: MoveToState ReadClient[0x72000840]: Moving to [AwaitingIn]
[1653405613.172170][8491:8492] CHIP:EM: Retransmitting MessageCounter:13517467 on exchange 52699i Send Cnt 1
[1653405613.173303][8491:8492] CHIP:IN: Sending encrypted msg 0x56440cc4 with MessageCounter:13517467 to 0x0000000000000457 (1) at monotonic time: 0000000000AC345B msec
[1653405613.174035][8491:8492] CHIP:CSM: FindOrEstablishSession: PeerId = D4183EC9FF3C7A41:0000000000000457
[1653405613.175462][8491:8492] CHIP:DIS: Resolving D4183EC9FF3C7A41:0000000000000457 ...
[1653405613.175910][8491:8492] CHIP:DL: Resolve type=_matter._tcp name=D4183EC9FF3C7A41-0000000000000457 interface=0
[1653405613.177582][8491:8492] CHIP:DL: Mdns : OnNewInterface hostname:5EC26475FB39E119.local. fullname:D4183EC9FF3C7A41-0000000000000457._matter._tcp.local. interface: 44
[1653405613.179755][8491:8492] CHIP:DL: Mdns: OnNewAddress interface: 44 ip:fd03:329c:d7fa:1:b37f:261a:d2b8:992a
[1653405613.180429][8491:8492] CHIP:DIS: Node ID resolved for D4183EC9FF3C7A41:0000000000000457
[1653405613.180650][8491:8492] CHIP:DIS: 	Hostname: 5EC26475FB39E119
[1653405613.180873][8491:8492] CHIP:DIS: 	IP Address #1: fd03:329c:d7fa:1:b37f:261a:d2b8:992a
[1653405613.181102][8491:8492] CHIP:DIS: 	Port: 5540
[1653405613.181297][8491:8492] CHIP:DIS: 	Mrp Interval idle: 5000 ms
[1653405613.181478][8491:8492] CHIP:DIS: 	Mrp Interval active: 2000 ms
[1653405613.182783][8491:8492] CHIP:DIS: Lookup clearing interface for non LL address
[1653405613.183269][8491:8492] CHIP:DIS: UDP:[fd03:329c:d7fa:1:b37f:261a:d2b8:992a%eth0]:5540: new best score: 6
[1653405613.183559][8491:8492] CHIP:DIS: Checking node lookup status after 8 ms
[1653405613.184251][8491:8492] CHIP:DIS: Keeping DNSSD lookup active
[1653405613.375368][8491:8492] CHIP:DIS: Checking node lookup status after 200 ms
[1653405613.376648][8491:8492] CHIP:CTL: Updating device address to UDP:[fd03:329c:d7fa:1:b37f:261a:d2b8:992a]:5540 while in state 5
[1653405613.377196][8491:8492] CHIP:DIS: Discovery does not require any more timeouts
[1653405619.148161][8491:8492] CHIP:EM: Retransmitting MessageCounter:13517467 on exchange 52699i Send Cnt 2
[1653405619.148630][8491:8492] CHIP:IN: Sending encrypted msg 0x56440cc4 with MessageCounter:13517467 to 0x0000000000000457 (1) at monotonic time: 0000000000AC4BB2 msec
[1653405619.963850][8491:8492] CHIP:DMG: Time out! failed to receive report data from Exchange: 52699i
[1653405619.964309][8491:8492] CHIP:DMG: Computing Resubscribe policy: attempts 8, max wait time 210000 ms, selected wait time 84593 ms
[1653405619.964421][8491:8492] CHIP:DMG: Will try to Resubscribe to 01:0000000000000457 at retry index 8 after 84593ms
[1653405619.964465][8491:8492] CHIP:DMG: MoveToState ReadClient[0x72000840]: Moving to [      Idle]
[1653405628.777165][8491:8492] CHIP:EM: Retransmitting MessageCounter:13517467 on exchange 52699i Send Cnt 3
[1653405628.777468][8491:8492] CHIP:IN: Sending encrypted msg 0x56440cc4 with MessageCounter:13517467 to 0x0000000000000457 (1) at monotonic time: 0000000000AC714F msec
[1653405643.467177][8491:8492] CHIP:EM: Retransmitting MessageCounter:13517467 on exchange 52699i Send Cnt 4
[1653405643.467620][8491:8492] CHIP:IN: Sending encrypted msg 0x56440cc4 with MessageCounter:13517467 to 0x0000000000000457 (1) at monotonic time: 0000000000ACAAB1 msec
[1653405668.390183][8491:8492] CHIP:EM: Failed to Send CHIP MessageCounter:13517467 on exchange 52699i sendCount: 4 max retries: 4
```

EFR32 device
```
<error > [IN] Data received on an unknown session (LSID=49596). Dropping it!
<detail> [IN] UDP Message Received packet nb : 37 SrcAddr : fd03:329c:d7fa:1:bc:a93a:44bb:fbff[42992] DestAddr : fd03:329c:d7fa:1:b37f:261a:d2b8:992a[5540] Payload Length 59
<error > [IN] Data received on an unknown session (LSID=49596). Dropping it!
<detail> [IN] UDP Message Received packet nb : 38 SrcAddr : fd03:329c:d7fa:1:bc:a93a:44bb:fbff[42992] DestAddr : fd03:329c:d7fa:1:b37f:261a:d2b8:992a[5540] Payload Length 59
<error > [IN] Data received on an unknown session (LSID=49596). Dropping it!
<detail> [IN] UDP Message Received packet nb : 39 SrcAddr : fd03:329c:d7fa:1:bc:a93a:44bb:fbff[42992] DestAddr : fd03:329c:d7fa:1:b37f:261a:d2b8:992a[5540] Payload Length 59
<error > [IN] Data received on an unknown session (LSID=49596). Dropping it!
<detail> [IN] UDP Message Received packet nb : 40 SrcAddr : fd03:329c:d7fa:1:bc:a93a:44bb:fbff[42992] DestAddr : fd03:329c:d7fa:1:b37f:261a:d2b8:992a[5540] Payload Length 59
<error > [IN] Data received on an unknown session (LSID=49596). Dropping it!
```

If we restart the Linux controller, then we can again subscribe and receive notifications as expected.

kghost, 2022-05-24:
This is a known issue, where the controller doesn't aware that the device has been rebooted.

kghost, 2022-05-24:
#17298 axiom 5 cover the scenario, but a solution is not proposed yet.

bzbarsky-apple, 2022-05-25:
Duplicate of #15766.

closed 2022-05-25 by bzbarsky-apple

yunhanw-google, 2022-05-25:
should duplicate of #16202 https://github.com/project-chip/connectedhomeip/issues/16202