project-chip/connectedhomeip #18764: Pairing SED fails

dtodor, 2022-05-24:
Pairing an EFR32 sleepy end device is not possible, the controller cannot find the operational device. This has been tested with a custom Linux controller as well as with the chip-tool.

Output chip-tool: 
[chip-tool.log](https://github.com/project-chip/connectedhomeip/files/8764548/chip-tool.log)

Output EFR32: 
[efr32-sed.log](https://github.com/project-chip/connectedhomeip/files/8764551/efr32-sed.log)

Pairing FTD works fine. The EFR32 lighting-app example has been built using the following command:
```
gn gen out/sleepy '--args=efr32_board="BRD4161A" enable_sleepy_device=true chip_openthread_ftd=false'
```

dtodor, 2022-05-24:
This EFR32 output seems weird (Device Role: DETACHED):
```
<detail> [DL] OpenThread State Changed (Flags: 0x1117d11d)
<detail> [DL]    Device Role: DETACHED
<detail> [DL]    Network Name: OpenThread-858c
<detail> [DL]    PAN Id: 0x858C
<detail> [DL]    Extended PAN Id: 0xF0203F8B900110FC
<detail> [DL]    Channel: 23
<detail> [DL]    Mesh Prefix: fd6a:2dcf:2b13:a3ef:0:0:0:0/64
<detail> [DL]    Thread Unicast Addresses:
<detail> [DL]         fd6a:2dcf:2b13:a3ef:a4e7:336c:333b:fff/64 valid
<detail> [DL]         fe80:0:0:0:f431:8af6:db0b:b855/64 valid preferred
```

This is the state of the Thread Border Router:
```
# ot-ctl 

> state
leader
Done
```

bzbarsky-apple, 2022-05-25:
@Damian-Nordic Any ideas what might be going on here?

jmartinez-silabs, 2022-05-25:
If this is the case there is a significant regression. SED build was tested and validated not long ago. I can't say I tried it lately. I will need to reproduce

woody-apple, 2022-05-27:
SDK Spec Review: We do not believe this to be a spec compliance issue, but just a platform bug.

jmartinez-silabs, 2022-06-03:
@dtodor Hi, Can you let me know what commit hash you used for your OTBR.

I tested it on my end with the current master. The SED was able to join the thread network correctly. Chip-tool did fail to discover the device a few times though which I don't usually see with non-SED builds. I was able to get it commissioned and send commands but I had to retry the commissioning twice before it got discovered correctly and the case session start. It does seem less stable than non-sed.

my OTBR is also on an old commit I will try to update that too.

dtodor, 2022-06-03:
This is the OTBR commit we are currently using: 1a424f3445ea2a7cf37ef8add3886cdfdd607472

dtodor, 2022-06-03:
I was also able to have the SED join the Thread network, but the CASE establishment never completed successfully.

mkardous-silabs, 2023-04-25:
Closing this issue since its a year old and EFR32 SED i have been tested since.

closed 2023-04-25 by mkardous-silabs