project-chip/connectedhomeip #18786: Android chip tool resolves the matter service with only IPV4 address

screensforme, 2022-05-25:
Android CHIP Tool resolves matter service  with only IPv4 address.

I provisioned chip device using wifi in android device, using IP address.
Then Light On/Off & Level Cluster -> UpdateAddress

Then I see android mdnssd resolves the matter service only on IPV4 even though it discovers both IPv4 & IPv6 address.
Could please check this. Is this any problem with android platform or android device configuration ?

Logs:
05-25 10:19:53.930 16116 16116 D NsdManagerServiceResolver: Starting service resolution for 'E4B75FF5A2A4E172-0000000000000005'
05-25 10:19:53.930  1225  1767 D NsdService: Resolve service
05-25 10:19:53.930  1225  1767 D NsdService: mdnssd [resolve, 27, E4B75FF5A2A4E172-0000000000000005, _matter._tcp, local.]
05-25 10:19:53.931   713   732 D MDnsDS  : resolveService(27, (null), E4B75FF5A2A4E172-0000000000000005, _matter._tcp, local.)
05-25 10:19:53.931   713   732 D MDnsDS  : startMonitoring 27
05-25 10:19:53.931   713   732 D MDnsDS  : triggering rescan
05-25 10:19:53.931   713   732 D MDnsDS  : resolveService successful
05-25 10:19:53.931   713   731 D MDnsDS  : Monitor poll got data pollCount = 2, 1
05-25 10:19:53.931   713   731 D MDnsDS  : controlSocket shows revent= 1
05-25 10:19:53.931   713   731 D MDnsDS  : MDnsSdListener::Monitor got 1
05-25 10:19:53.931   713   731 D MDnsDS  : MDnsSdListener::Monitor poll rescanning - size=10, live=2
05-25 10:19:53.931   713   731 D MDnsDS  : Going to poll with pollCount 3
05-25 10:19:54.209   713   731 D MDnsDS  : Monitor poll got data pollCount = 3, 1
05-25 10:19:54.209   713   731 D MDnsDS  : Monitor found [1].revents = 1 - calling ProcessResults
05-25 10:19:54.209   713   731 D MDnsDS  : resolve succeeded for 27 finding E4B75FF5A2A4E172-0000000000000005._matter._tcp.local. at 240AC4EE3854.local.:5540 with txtLen 21
05-25 10:19:54.209   713   731 D MDnsDS  : controlSocket shows revent= 0
05-25 10:19:54.209   713   731 D MDnsDS  : Going to poll with pollCount 3
05-25 10:19:54.210  1225  1767 D NsdService: Native daemon message SERVICE_RESOLVED: 608 27 "E4B75FF5A2A4E172-0000000000000005._matter._tcp.local." "240AC4EE3854.local." 5540 21 "CENSST01MDAwB0NSQT0zMDADVD0x"
05-25 10:19:54.211  1225  1767 D NsdService: mdnssd [stop-resolve, 27]
05-25 10:19:54.211   713   732 D MDnsDS  : Stopping resolve with ref 0x78704175e8
05-25 10:19:54.211   713   732 D MDnsDS  : freeServiceRef 27
05-25 10:19:54.211   713   732 D MDnsDS  : triggering rescan
05-25 10:19:54.211  1225  1767 D NsdService: mdnssd [getaddrinfo, 28, 240AC4EE3854.local.]
05-25 10:19:54.212   713   731 D MDnsDS  : Monitor poll got data pollCount = 3, 2
05-25 10:19:54.212   713   732 D MDnsDS  : getAddrInfo(28, (null) 0, 240AC4EE3854.local.)
05-25 10:19:54.212   713   731 D MDnsDS  : Monitor found [1].revents = 32 - calling ProcessResults
05-25 10:19:54.212   713   731 D MDnsDS  : controlSocket shows revent= 1
05-25 10:19:54.212   713   731 D MDnsDS  : MDnsSdListener::Monitor got 1
05-25 10:19:54.212   713   731 D MDnsDS  : MDnsSdListener::Monitor poll rescanning - size=10, live=1
05-25 10:19:54.212   713   731 D MDnsDS  : Going to poll with pollCount 2
05-25 10:19:54.212   713   732 D MDnsDS  : startMonitoring 28
05-25 10:19:54.212   713   732 D MDnsDS  : triggering rescan
05-25 10:19:54.212   713   732 D MDnsDS  : getAddrInfo successful
05-25 10:19:54.212   713   731 D MDnsDS  : Monitor poll got data pollCount = 2, 1
05-25 10:19:54.213   713   731 D MDnsDS  : controlSocket shows revent= 1
05-25 10:19:54.213   713   731 D MDnsDS  : MDnsSdListener::Monitor got 1
05-25 10:19:54.213   713   731 D MDnsDS  : MDnsSdListener::Monitor poll rescanning - size=10, live=2
05-25 10:19:54.213   713   731 D MDnsDS  : Going to poll with pollCount 3
05-25 10:19:54.213   713   731 D MDnsDS  : Monitor poll got data pollCount = 3, 1
05-25 10:19:54.213   713   731 D MDnsDS  : Monitor found [1].revents = 1 - calling ProcessResults
05-25 10:19:54.213   713   731 D MDnsDS  : getAddrInfo succeeded for 28: 28 "240AC4EE3854.local." 120 192.168.1.25
05-25 10:19:54.213   713   731 D MDnsDS  : getAddrInfo succeeded for 28: 28 "240AC4EE3854.local." 120 fe80::260a:c4ff:feee:3854
05-25 10:19:54.213   713   731 D MDnsDS  : getAddrInfo succeeded for 28: 28 "240AC4EE3854.local." 120 2401:4900:1f25:7281:260a:c4ff:feee:3854
05-25 10:19:54.213   713   731 D MDnsDS  : controlSocket shows revent= 0
05-25 10:19:54.213   713   731 D MDnsDS  : Going to poll with pollCount 3
05-25 10:19:54.213  1225  1767 D NsdService: Native daemon message SERVICE_GET_ADDR_SUCCESS: 612 28 "240AC4EE3854.local." 120 192.168.1.25
05-25 10:19:54.214  1225  1767 D NsdService: mdnssd [stop-getaddrinfo, 28]
05-25 10:19:54.214   713   732 D MDnsDS  : Stopping getaddrinfo with ref 0x78704175b8
05-25 10:19:54.214   713   732 D MDnsDS  : freeServiceRef 28
05-25 10:19:54.214   713   732 D MDnsDS  : triggering rescan
05-25 10:19:54.215   713   731 D MDnsDS  : Monitor poll got data pollCount = 3, 2
05-25 10:19:54.215   713   731 D MDnsDS  : Monitor found [1].revents = 32 - calling ProcessResults
05-25 10:19:54.215   713   731 D MDnsDS  : controlSocket shows revent= 1
05-25 10:19:54.215  1225  1767 E NsdService: id 28 for SERVICE_GET_ADDR_SUCCESS has no client mapping
05-25 10:19:54.215   713   731 D MDnsDS  : MDnsSdListener::Monitor got 1
05-25 10:19:54.215   713   731 D MDnsDS  : MDnsSdListener::Monitor poll rescanning - size=10, live=1
05-25 10:19:54.215   713   731 D MDnsDS  : Going to poll with pollCount 2
05-25 10:19:54.215  1225  1767 E NsdService: Unhandled { when=-1ms what=393242 obj=com.android.server.NsdService$NativeEvent@3104a54 target=com.android.internal.util.StateMachine$SmHandler }
05-25 10:19:54.215  1225  1767 E NsdService: id 28 for SERVICE_GET_ADDR_SUCCESS has no client mapping
05-25 10:19:54.215  1225  1767 E NsdService: Unhandled { when=-1ms what=393242 obj=com.android.server.NsdService$NativeEvent@58d6dfd target=com.android.internal.util.StateMachine$SmHandler }
05-25 10:19:54.215 16116 17047 I NsdManagerServiceResolver: Resolved service 'E4B75FF5A2A4E172-0000000000000005' to /192.168.1.25

screensforme, 2022-06-15:
Hello @g-coppock 
I noted your PR https://github.com/project-chip/connectedhomeip/pull/19394 related to DNS-SD & IPV6 for android.

I am stuck with the above problem in my environment. And my android chiptool is not able to resolve the matter device(m5stack) with IPV6 address. Did you not faced above issue ? How can we test android chip tool using IPV6, is there any extra configuration is needed on device side.

I'd really appreciate your help on this.

g-coppock, 2022-06-15:
Hello, we are using an internal solution for mDNS rather than `NsdManager` so I can't suggest directly. However, you may want to try disabling IPv4 support on the device (e.g. for the `examples` apps, `idf.py menuconfig` -> `Compoment config > CHIP core > General Options > Disable IPv4 functionality in the CHIP stack`) and see if that helps.

screensforme, 2022-06-16:
Thanks for your response. I shall try this.

@xylophone21 Since you are primary author of NsdManagerServiceResolver.java, I am asking this query.
Is the above mentioned problem is known in android ? 
When there is both IPv4 and IPv6 address, NsdManagerServiceResolver will not be able to resolve IPv6 address ?

andy31415, 2022-06-16:
In other implementations (e.g. avahi), the resolver often returns the first IP address it sees. The logic being "server provides a list of available ip addresses, all of them being equivalent".

As a result, it seems that if a device is able to advertise both ipv6 and v4, any could be used. I would suggest to try with something that is never ipv4 like thread devices, to make sure things work correctly.

tonapi-abhi, 2022-09-27:
This issue still exists, there is no provision in _NsdManager_ to resolve service based on IP Address Type i.e to resolve services under same name with specified Address IPv6 or IPv4 address Type. The Chip tool still resolves IPv4 address even though it discovers both. 
One possible workaround is to publish two services itself from the device side with different names each with Ipv4 and Ipv6 address types, and then to only resolve the service with Ipv6 naming in the tool side. This works but I am not sure if this is the correct way to proceed..
Can anyone confirm this, i am open to suggestions. You can find the logs below..

-Before Workaround

- Device Logs:- 
I (48343) chip[ZCL]: OpCreds: ACL entry created for Fabric index 0x1 CASE Admin Subject 0x000000000001B669
I (48343) chip[DIS]: Advertise operational node A6EC5EBC94DCB11F-0000000000000001
I (48363) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'; instance name: A6EC5EBC94DCB11F-0000000000000001.
**I (48373) chip[DIS]: mDNS service published: _matter._tcp**
I (48373) chip[ZCL]: OpCreds: successfully created fabric index 0x1 via AddNOC
I (48393) chip[EM]: <<< [E:39910r M:262123077 (Ack:191747213)] (S) Msg TX to 1:FFFFFFFB00000000 [B11F] --- Type 0001:09 (IM:InvokeCommandResponse)
I (48403) chip[IN]: (S) Sending msg 262123077 on secure session with LSID: 58230

- ChipTool Logs:-
09-26 16:57:16.336   465  1081 D MDnsDS  : Starting MDNSD
09-26 16:57:26.163 30399 30583 D DIS     : Resolving A6EC5EBC94DCB11F:0000000000000001 ...
**09-26 16:57:26.198 30399 30583 D NsdManagerServiceResolver: resolve: Starting service resolution for 'A6EC5EBC94DCB11F-0000000000000001' type '_matter._tcp'**
09-26 16:57:26.365 30399 30583 D DIS     : Checking node lookup status after 202 ms
**09-26 16:57:26.371 30399 30580 I NsdManagerServiceResolver: Resolved service 'A6EC5EBC94DCB11F-0000000000000001' to /192.168.96.20**
09-26 16:57:26.391   948  1195 E NsdService: Unique id with no client mapping: 40
09-26 16:57:26.391   948  1195 E NsdService: Unhandled { when=-27ms what=393242 obj=com.android.server.NsdService$NativeEvent@6ba8cf0 target=com.android.internal.util.StateMachine$SmHandler }
09-26 16:57:26.424 30399 30580 D DIS     :  ----- entry [0] : T 1
09-26 16:57:26.430 30399 30580 D DIS     :  ----- entry [1] : SAI 300
09-26 16:57:26.432 30399 30580 D DIS     :  ----- entry [2] : SII 5000
09-26 16:57:26.433 30399 30580 D DIS     : Node ID resolved for A6EC5EBC94DCB11F:0000000000000001
09-26 16:57:26.433 30399 30580 D DIS     : Hostname: 192.168.96.20
09-26 16:57:26.433 30399 30580 D DIS     : IP Address 1: 192.168.96.20
09-26 16:57:26.433 30399 30580 D DIS     : Port: 5540
09-26 16:57:26.433 30399 30580 D DIS     : Mrp Interval idle: 5000 ms
09-26 16:57:26.434 30399 30580 D DIS     : Mrp Interval active: 300 ms
09-26 16:57:26.434 30399 30580 D DIS     : TCP Supported: 1
09-26 16:57:26.434 30399 30580 D DIS     : Lookup clearing interface for non LL address
09-26 16:57:26.434 30399 30580 D DIS     : UDP:192.168.96.20:5540: new best score: 2
09-26 16:57:26.434 30399 30580 D DIS     : Checking node lookup status after 271 ms
09-26 16:57:26.449 30399 30580 D DIS     : OperationalSessionSetup[1:0000000000000001]: Updating device address to UDP:192.168.96.20:5540 while in state 2

-After Workaround

- Device Logs:-
Validating NOC chain
I (121253) chip[FP]: NOC chain validation successful
I (121253) chip[FP]: Added new fabric at index: 0x1
I (121263) chip[FP]: Assigned compressed fabric ID: 0x839D292B139DC14C, node ID: 0x0000000000000002
I (121273) chip[TS]: Last Known Good Time: 2022-09-22T09:46:56
I (121283) chip[TS]: New proposed Last Known Good Time: 2021-06-10T00:00:00
I (121283) chip[TS]: Retaining current Last Known Good Time
I (121313) chip[ZCL]: OpCreds: ACL entry created for Fabric index 0x1 CASE Admin Subject 0x000000000001B669
I (121313) chip[DIS]: Advertise operational node 839D292B139DC14C-0000000000000002
**I (121333) chip[DIS]: mDNS service published: _matter_IPv4._tcp
I (121353) chip[DIS]: mDNS service published: _matter_IPv6._tcp**
I (121353) chip[ZCL]: OpCreds: successfully created fabric index 0x1 via AddNOC
I (121363) chip[EM]: <<< [E:46923r M:86006312 (Ack:252699012)] (S) Msg TX to 1:FFFFFFFB00000000 [C14C] --- Type 0001:09 (IM:InvokeCommandResponse)
I (121373) chip[IN]: (S) Sending msg 86006312 on secure session with LSID: 27881
E (121383) chip[DL]: Long dispatch time: 812 ms, for event type 3

- ChipTool Logs:-
09-26 15:53:55.871 25080 25080 I SA      : [OR] onReceive log=[SA = 2.2.03.98 V = 25 HWD = 1920X1080 2.625 dpi = 420  SIZE = 2 LOCALE = en_GB CSC = XXV MCC = 0 MNC 0 T = eng DEVICE = jadelte P = jadeltedx I = NMF26X M = SM-C710F OKLEFT false DIS jadeltedx-eng 7.1.1 NMF26X C710FDXE0AQG5 test-keys PSS = 5.460701736796467  ]
09-26 15:54:27.969   465  1081 D MDnsDS  : Starting MDNSD
09-26 15:54:35.462 25108 25934 D DIS     : Resolving 839D292B139DC14C:0000000000000002 ...
**09-26 15:54:35.473 25108 25934 D NsdManagerServiceResolver: resolve: Starting service resolution for '839D292B139DC14C-0000000000000002' type '_matter_IPv6._tcp'
09-26 15:54:35.501 25108 25921 I NsdManagerServiceResolver: Resolved service '839D292B139DC14C-0000000000000002' to /fe80::7a21:84ff:fe93:3c28**
09-26 15:54:35.503 25108 25921 D DIS     :  ----- entry [0] : T 1
09-26 15:54:35.503 25108 25921 D DIS     :  ----- entry [1] : SAI 300
09-26 15:54:35.503 25108 25921 D DIS     :  ----- entry [2] : SII 5000
[... cut ...]
09-26 15:54:35.503 25108 25921 D DIS     : UDP:[fe80::7a21:84ff:fe93:3c28]:5540: new best score: 3
09-26 15:54:35.503 25108 25921 D DIS     : Checking node lookup status after 41 ms
09-26 15:54:35.504 25108 25921 D DIS     : Keeping DNSSD lookup active
09-26 15:54:35.663 25108 25934 D DIS     : Checking node lookup status after 200 ms
09-26 15:54:35.663 25108 25934 D DIS     : OperationalSessionSetup[1:0000000000000002]: Updating device address to UDP:[fe80::7a21:84ff:fe93:3c28]:5540 while in state 2

closed 2023-05-01 by stale[bot]