project-chip/connectedhomeip #18789: Support mDNSResponder on Linux (feature request)

dtodor, 2022-05-25:
The platform mDNS implementation (i.e. `chip_mdns = "platform"`) on Linux uses Avahi. The OpenThread Border Router on the other hand supports both Avahi and the mDNSResponder. It would be favourable for systems running an OTBR and a Matter controller to use the same mDNS daemon, which currently is only possible with Avahi.

Adding support for the mDNSResponder to the Linux platform is straight-forward - the Darwin implementation requires minimal changes:
```
diff --git a/src/platform/Darwin/DnssdContexts.cpp b/src/platform/Linux/DnssdContexts.cpp
index 2699bec21..18c60c377 100644
--- a/src/platform/Darwin/DnssdContexts.cpp
+++ b/src/platform/Linux/DnssdContexts.cpp
@@ -143,6 +143,16 @@ MdnsContexts::~MdnsContexts()
     }
 }
 
+void MdnsWatchCallback(chip::System::SocketEvents events, intptr_t data)
+{
+    DNSServiceRef sdRef = reinterpret_cast<DNSServiceRef>(data);
+    auto err            = DNSServiceProcessResult(sdRef);
+    if (err != kDNSServiceErr_NoError)
+    {
+        ChipLogError(DeviceLayer, "Failed to process mdns result (%s)", Error::ToString(err));
+    }
+}
+
 CHIP_ERROR MdnsContexts::Add(GenericContext * context, DNSServiceRef sdRef)
 {
     VerifyOrReturnError(context != nullptr, CHIP_ERROR_INVALID_ARGUMENT);
@@ -153,13 +163,32 @@ CHIP_ERROR MdnsContexts::Add(GenericContext * context, DNSServiceRef sdRef)
         return CHIP_ERROR_INVALID_ARGUMENT;
     }
 
-    auto err = DNSServiceSetDispatchQueue(sdRef, chip::DeviceLayer::PlatformMgrImpl().GetWorkQueue());
-    if (kDNSServiceErr_NoError != err)
+    auto fd = DNSServiceRefSockFD(sdRef);
+    if (fd == -1)
     {
         chip::Platform::Delete(context);
         return CHIP_ERROR_INTERNAL;
     }
 
+    auto err = DeviceLayer::SystemLayerSockets().StartWatchingSocket(fd, &context->socketWatch);
+    if (CHIP_NO_ERROR != err)
+    {
+        chip::Platform::Delete(context);
+        return err;
+    }
+    err = DeviceLayer::SystemLayerSockets().SetCallback(context->socketWatch, MdnsWatchCallback, reinterpret_cast<intptr_t>(sdRef));
+    if (CHIP_NO_ERROR != err)
+    {
+        chip::Platform::Delete(context);
+        return err;
+    }
+    err = DeviceLayer::SystemLayerSockets().RequestCallbackOnPendingRead(context->socketWatch);
+    if (CHIP_NO_ERROR != err)
+    {
+        chip::Platform::Delete(context);
+        return err;
+    }
+
     context->serviceRef = sdRef;
     mContexts.push_back(context);
 
@@ -179,6 +208,8 @@ CHIP_ERROR MdnsContexts::Remove(GenericContext * context)
             continue;
         }
 
+        DeviceLayer::SystemLayerSockets().StopWatchingSocket(&context->socketWatch);
+
         Delete(*iter);
         mContexts.erase(iter);
         found = true;
@@ -352,7 +383,6 @@ CHIP_ERROR ResolveContext::OnNewLocalOnlyAddress()
 {
     sockaddr_in6 sockaddr;
     memset(&sockaddr, 0, sizeof(sockaddr));
-    sockaddr.sin6_len    = sizeof(sockaddr);
     sockaddr.sin6_family = AF_INET6;
     sockaddr.sin6_addr   = in6addr_loopback;
     sockaddr.sin6_port   = htons((unsigned short) interfaces[kDNSServiceInterfaceIndexLocalOnly].service.mPort);
diff --git a/src/platform/Darwin/DnssdImpl.h b/src/platform/Linux/DnssdImpl.h
index e41b61421..bbacaf167 100644
--- a/src/platform/Darwin/DnssdImpl.h
+++ b/src/platform/Linux/DnssdImpl.h
@@ -39,6 +39,7 @@ struct GenericContext
     ContextType type;
     void * context;
     DNSServiceRef serviceRef;
+    chip::System::SocketWatchToken socketWatch;
 
     virtual ~GenericContext() {}
```

If supporting the mDNSResponder on Linux is a viable scenario, we could provide a merge request.

andy31415, 2022-05-25:
@dtodor - if we were to support mDNSResponder, I think it should be a option like chip_mdns supporting "minimal", "platform" and "responder" (or some other variation of names).

Need to make it configurable, beyond that having more options to testing things is great. We also need some maintenance for it especially at start. Would you be willing to fix issues if we find them to be mDNSResponder specific?

dtodor, 2022-05-26:
I will discuss this with our client and will come back to you later. We are already doing the alignment with the Darwin implementation on a regular basis.

closed 2022-12-03 by stale[bot]