project-chip/connectedhomeip #18790: [TC-IDM-4.3]while verifying 2nd subscription with dataversion set, DUT Fails to sends a report data action for the attribute value,

Rajashreekalmane, 2022-05-25:
#### Problem
As per test-plan the test-step 19 says, 
[image]
currently while verifying 2nd subscription with data version set received earlier, DUT Fails to sends a report data action for the attribute value.

 * expected behavior:-DUT should send a report data action for the attribute value.
 * actual behavior:-Report data consists only subscriptionID.
 
[image]

 * steps to reproduce
  ./chip-tool interactive start
onoff subscribe on-off 100 1000 1 1
 onoff subscribe on-off 100 1000 1 1 --data-version  0xbf2fc26d 

system configuration:-
Chip-Tool: RPI(chip-tool)
DUT: RPI(all-clusters-app)
Platform : IP
Commit-ID: https://github.com/project-chip/connectedhomeip/commit/7afd35229da424a139642e4ebbe68bb4d71554cf
Platform:-Linux platform

Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-4-3-report-data-messages-post-subscription-activation-from-dut-test-cases

PFA log:-
[IDM-4.3 19th step.txt](https://github.com/project-chip/connectedhomeip/files/8769952/IDM-4.3.19th.step.txt)

bzbarsky-apple, 2022-05-25:
The whole point of sending a data version in a subscribe is that if the version matches the value does not need to be sent.

bzbarsky-apple, 2022-05-25:
@cjandhyala @mrjerryjohns @yunhanw-google Sounds like the test plan needs fixing.

closed 2022-05-25 by bzbarsky-apple