project-chip/connectedhomeip #18791: [TE9] Idm-1.2 scan-network command doesnot show multiple networks and fails in last step

PSONALl, 2022-05-25:
#### Problem
For test case: TC-IDM-1.2:
 * expected behavior : after sending Scan networks command, DUT sends back a response which is of data type list. and list down all the networks
 * actual behavior : scan-networks command only shows one entry in list . command fails in last step of the testcase with error: Run command failure: IM Error 0x00000585: General error: 0x85 (INVALID_COMMAND) 
 * system configuration :
        -  DUT used : ESP32 
        -  app use: all-clusters-app
        - chip-tool and DUT commit id : cfc35951be66a664a6efdadea56d1b8ea6e63e96  (TE9 commit id)
    
 *  logs
        - esp32 : [esp32.txt](https://github.com/project-chip/connectedhomeip/files/8770368/esp32.txt)
        - chip-tool : [th.txt](https://github.com/project-chip/connectedhomeip/files/8770370/th.txt)

mrjerryjohns, 2022-05-26:
Could you read out the `acceptedCommandList` from the network commissioning cluster? I suspect that the wifi interface is somehow not being enabled in the cluster, which might explain the error of 'INVALID_COMMAND'.

shubhamdp, 2022-06-17:
Tested on Commit: 26d6c544ae6e98034be6e2f865961a6df71d9429, I can see the list of networks, please see the attached logs.
[scan_network.log](https://github.com/project-chip/connectedhomeip/files/8925037/scan_network.log)

PSONALl, 2022-06-17:
> Tested on Commit: [26d6c54](https://github.com/project-chip/connectedhomeip/commit/26d6c544ae6e98034be6e2f865961a6df71d9429), I can see the list of networks, please see the attached logs. [scan_network.log](https://github.com/project-chip/connectedhomeip/files/8925037/scan_network.log)

closing this issue

closed 2022-06-17 by PSONALl