project-chip/connectedhomeip #18793: TC-WNCV-4.1 - Liftpercent 100th's value is not matched with expected outcome

somu1710, 2022-05-25:
**Problem:**

In Windowcovering cluster [TC-WNCV-4.1] - The value of liftpercent 100th's is not matched with the expected outcome and current position lift percentage has shows "0" as value instead of "25".

**Expected outcome:**

While we giving the lift percentage value as 25 , Liftpercent 100ths value should be 2500
**Commands used:**
1../chip-tool windowcovering go-to-lift-percentage 25 1 1
                       status = 0x00 (SUCCESS)
2../chip-tool windowcovering read target-position-lift-percent100ths 1 1
                    TargetPositionLiftPercent100ths: 2500

**Actual outcome:**
But while  giving the lift percentage value as 25, It gives 25 as the  Liftpercent 100ths value 
**Commands used:**
1../chip-tool windowcovering go-to-lift-percentage 25 1 1
                       status = 0x00 (SUCCESS)
2../chip-tool windowcovering read target-position-lift-percent100ths 1 1
                      TargetPositionLiftPercent100ths: 25

**Expected outcome:**
 While reading the current-position-lift-percentage ,it should show 25 as output but it shows "0"
**Commands used:**
 ./chip-tool  windowcovering read current-position-lift-percentage 1 1
           CurrentPositionLiftPercentage: 25 

**Actual outcome:**
**Commands used:**
 ./chip-tool  windowcovering read current-position-lift-percentage 1 1
             CurrentPositionLiftPercentage: 0

App used:" All-clusters-app"
Platform:"Raspberrypi"
Commit used: 7afd35229da424a139642e4ebbe68bb4d71554cf
 
Steps: 2a,2c,3c,3d,4a,4c,5c,5d
    
For reference here i attached the screen shot and logs:

[image]

[WNCV-4.1 - 100THS value fail.txt](https://github.com/project-chip/connectedhomeip/files/8770507/WNCV-4.1.-.100THS.value.fail.txt)

jmeg-sfy, 2022-05-25:
we changed this command recently
no more optional always input a 100th percent value

somu1710, 2022-05-25:
Ok @jmeg-sfy

closed 2022-05-25 by somu1710