project-chip/connectedhomeip #18805: How to get a notifications when a device appear online and when it seems disconnected ?

caipiblack, 2022-05-25:
#### Problem

I am building a Matter controller using TE9 commit. Currently with this controller we can commission and control devices.

To improve the user experience, we would like to be able to detect when a device appears online and when a device seems "offline" (not speaking since long time).

**Currently I'm not sure about how to do it with the Matter stack and if this functionality is supported or not.**

Currently I found this:

```
chip::Controller::DeviceDiscoveryDelegate:
void OnDiscoveredDevice(const chip::Dnssd::DiscoveredNodeData & nodeData)
```

This function is not what i am looking for as we don't have the NodeID information in nodeData. And I'm not sure if I can use it in another context than "pairing".

Also I found this:

```
NodeListener (src/lib/address_resolve/AddressResolve.h):
    /// Callback executed once only for a lookup, when the final address of a
    /// node is considered to be the best choice for reachability.
    ///
    /// The callback is expected to be executed within the CHIP event loop
    /// thread.
    virtual void OnNodeAddressResolved(const PeerId & peerId, const ResolveResult & result) = 0;

    /// Node resolution failure - occurs only once for a lookup, when an address
    /// could not be resolved - generally due to a timeout or due to DNSSD
    /// infrastructure returning an error.
    ///
    /// The callback is expected to be executed within the CHIP event loop
    /// thread.
    virtual void OnNodeAddressResolutionFailed(const PeerId & peerId, CHIP_ERROR reason) = 0;
```

But again, I'm not sure about how to use it, and if I can use it (as it is already used by `src/lib/address_resolve/AddressResolve_DefaultImpl.cpp`) (and it's possibly only for the "stack usage").

Any ideas ?

tcarmelveilleux, 2022-05-25:
The correct method is not discovery, since it is liable to TTL caching, especially for Thread devices. The correct method is to establish a subscription to devices you care to have their liveness information, and use the time-out of the subscription to trigger determining if still alive

caipiblack, 2022-05-27:
I understand, and my objective is also to use the subscribe, but for example:

Currently on TE9 with lighting examples for ESP32-C3, **after the comissioning succeed** it happens that the device loose the WiFi connection and get some difficulties to return on the network, so as we send a subscribe after commissioning succeed and the device loose WiFi at this point, we get a timeout.

**After that**, when you know the device is "offline", what is the strategy to check his presence ?
- Send a subscribe command every X minutes until success ?
- Wait for a message that notify about the device presence in the network and then send the subscribe ?

For the second statement we need to get a notification from the stack to implement it.

NOTE: The problem described on the example is possibly due to a bug in the lighting example on ESP32-C3 (I think it's when BLE is disabled we loose the WiFi) but it's not really important as our software need to be resistant against this.

closed 2022-12-03 by stale[bot]

LucaFagan, 2023-07-12:
I too am interested in this issues. Has anyone found a solution?

yunhanw-google, 2023-07-27:
controller can create a wildcard subscription with target device, if device goes offline, after max period(set by subscription), the controller would not receive any report from target, https://github.com/project-chip/connectedhomeip/blob/master/src/app/ReadClient.cpp#L634 would be called, the android chip-tool, for example, would get the corresponding callback to let application know the device goes "offline", then controller would use default fibonacci algorithm to retry the subscription, once the connection is up, the application received subscriptionEstablished callback(https://github.com/project-chip/connectedhomeip/blob/master/src/app/ReadClient.h#L138), which can be used as "Online" sign. thanks