project-chip/connectedhomeip #18836: TC-SU-2.2 - DUT  sends query image after 120 seconds instead of 180 seconds using busy command

somu1710, 2022-05-26:
Problem:
As per testplan [TC-SU-2.5] - step no.5 DUT sends queryimage set as busy for 180 seconds  but it sends after 120 seconds.

Solution:
 DUT needs to send the queryimage command after 3 mins(180 seconds) not to send after 2 mins (120 sec) as per spec.

CHIP:SWU: QueryImageResponse:
 CHIP:SWU:   status: 1
 CHIP:SWU:   delayedActionTime: 180 seconds
CHIP:SWU:   userConsentNeeded: 0

Platform- Raspberrypi
App used - All clusters app

Here i attach the logs for reference.
[OTA-PROVIDER PT.txt](https://github.com/project-chip/connectedhomeip/files/8777114/OTA-PROVIDER.PT.txt)

[OTA REQ 120S.txt](https://github.com/project-chip/connectedhomeip/files/8777116/OTA.REQ.120S.txt)

somu1710, 2022-05-26:
#https://github.com/project-chip/connectedhomeip/pull/18832

andy31415, 2022-05-26:
How is this bug report relating to #18616? Seems like both are regarding the same test case and complain about timeouts.

somu1710, 2022-05-26:
@andy31415 This issue is not related to this because #18616 issue is for update not available command it explains that the image should query after 2 mins but its not querying and this issue is related to busy command it querying after 2 mins instead of 3 mins as per test plan.

isiu-apple, 2022-05-26:
@somu1710, this issue looks identical to https://github.com/project-chip/connectedhomeip/issues/18432, and the fix was merged earlier today via https://github.com/project-chip/connectedhomeip/pull/18832.  Can you please re-test with this fix and dup the issues if identical?

somu1710, 2022-05-26:
@isiu-apple  Sure will check  with the latest SHA and close the ticket as per. I think its same issue

closed 2022-05-27 by somu1710